GeoscienceAustralia/ginan #8: -DENABLE_MONGODB=TRUE is default?

mattkingncl, 2022-03-21:
hi team

in cmake I found I needed to explicitly set -DENABLE_MONGODB=FALSE to avoid this dependency.

Doc suggests that " If you wish to enable the optional MONGO DB utilities you will need to add the -DENABLE_MONGODB=TRUE flag" which seems the opposite to my experience.

Also, on the github build page, could do with a better formatting on teh below:

_Since Ginan v1.2-alpha both gcc and g++ of version 9 are required, so make sure to update the gcc/g++ alternatives prior to compilation: sudo add-apt-repository ppa:ubuntu-toolchain-r/test sudo apt update sudo apt install gcc-9 g++-9 sudo update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-9 51 sudo update-alternatives --install /usr/bin/g++ g++ /usr/bin/g++-9 51_

matt

smcclusky, 2022-03-21:
Hi Matt,

With the Ginan v1.3-Beta release we have implemented a new algorithm that uses MongoDB to manage SSR RTCM3 corrections. Ie Ginan writes all corrections to the MongoDB before they are uploaded to the Ntrip Castor.
That is why the CMakeList.txt now has: option(ENABLE_MONGODB           "ENABLE_MONGODB"            ON) by default.
So you if want to build without Mongo you must explicitly turn it off. We are fixing the ReadMe.md file now to make this clear to users. 
Turning off Mongo however disables some of the DB related features of Ginan including uploading RTCM3 streams to an Ntrip Castor and exporting PEA output to MongoDB for analysis in the GinanEDA found in the /scripts/GinanEDA directory.

--Simon

mattkingncl, 2022-03-21:
Hi Simon

Thanks. So if I just want to do station positions, do I need MongoDB?

Matt King
Director of the Australian Centre for Excellence in Antarctic Science
School of Geography, Planning, and Spatial Sciences, University of Tasmania

This email is confidential, and is for the intended recipient only. Access, disclosure, copying, distribution, or reliance on any of it by anyone outside the intended recipient organisation is prohibited and may be a criminal offence. Please delete if obtained in error and email confirmation to the sender. The views expressed in this email are not necessarily the views of the University of Tasmania, unless clearly intended otherwise.

smcclusky, 2022-03-21:
No, to get the file based station coordinates (in .snx or .trace format) you don't (shouldn't) need MongoDB.

closed 2022-03-21 by mattkingncl