GeoscienceAustralia/ginan #9: Can not install MongDB

fencun, 2022-04-30:
CMake Error: cmake_symlink_library: System Error: Operation not supported
CMake Error: cmake_symlink_library: System Error: Operation not supported
src/libbson/CMakeFiles/bson_shared.dir/build.make:445: recipe for target 'src/libbson/libbson-1.0.so.0.0.0' failed
make[2]: *** [src/libbson/libbson-1.0.so.0.0.0] Error 1
make[2]: *** delecting “src/libbson/libbson-1.0.so.0.0.0”
CMakeFiles/Makefile2:405: recipe for target 'src/libbson/CMakeFiles/bson_shared.dir/all' failed
make[1]: *** [src/libbson/CMakeFiles/bson_shared.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2

smcclusky, 2022-05-09:
Can you please provide some more information about the system you are attempting to build Ginan on?

JohnDonovanGA, 2023-05-30:
No further info provided. Closing

closed 2023-05-30 by JohnDonovanGA