GeoscienceAustralia/ginan #10: Various compilation problems...

vallis, 2022-05-05:
This is on MacOS 12, but the problems seem more general:

- MongoDB and OpenMP are supposed to be optional, but they are not really. Even if ENABLE_MONGODB and ENABLE_PARALLELISATION are false, various source files contain code that requires them to be installed, and CMakeLists.txt still adds the corresponding libraries, so compilation fails.
- add_executable(peaUploader) seems to lack many sources files, resulting in linking errors. Adding ephemeris.cpp (which seems needed) results in a duplicate definition of ecef2rac.
- ${OPENSSL_INCLUDE_DIR} is needed in target_include_directories(...)
- On recent compilers, common/orbits.cpp needs to include <cmath> and use std::isnan (or perhaps include <math.h>, I'm not sure); common/rinexObsWrite.cpp needs to have "unsigned int" instead of "uint".

I may be able to PRs for the last two, but the first two seem beyond the scope of external review.

smcclusky, 2022-05-09:
Hi Vallis, Apologies for these problems you have encountered. 
We are currently working on a Ginan 1.3 Beta patch which is due to be released at the end of this week.
This patch should resolve all of the configuration and build problems you have encountered on you system.

vallis, 2022-05-23:
Thanks Simon! No pressure, but do you have an ETA on this? (So I can plan for some time to resume my experimentation...)

smcclusky, 2022-05-24:
Hi Vallis, Sorry for the delay, but the good news is, the 1.4-beta update release is coming in the next few hours. 
I hope we have addressed all of your issues, but we don't actually have an M1 Mac running OSX12 for testing so we can't be 100% sure.  I will be interested to get your feedback if there are remaining problems.
Thanks for your patience.

vallis, 2022-05-26:
Thanks Simon. I cloned v1.4-beta, but it seems to be incomplete... for instance, it's missing CMakeLists-C.txt and CMakeLists-F.txt. Do I understand correctly that now mongodb is a requirement? (ENABLE_MONGODB has disappeared from all source files.)

In the meantime, I gave another try to v1.3-beta, where I again have problems with excluding mongo code (I should really install it, right?). Also there may be something strange with Eigen: eigenIncluder.hpp tries to use Eigen::Vector as a template, but it can't do that with my Eigen (3.3.8) or according to the documentation. Also Eigen::placeholders::all doesn't seem to exist...

smcclusky, 2022-05-31:
Hi Vallis,

We have just pushed an update to Ginan 1.4-beta (1.4.1-beta). The problem with the CMakeList.txt file in the root directory has been rectified.

We have done some more testing of the required Ginan Boost, Eigen and  MongoDB dependencies. The README.md file has been updated accordingly. You will note that Eigen-3.4.0 is now the minimum requirement. (3.3.8 and 3.3.9 are not compatible).

You are correct that ENABLE_MONGODB has been removed and the installing MONGO_cxx driver dependency is now required, however starting and using MONGODB service is not required unless you want to use MONGODB to store output.

I will note that we have not built or tested a Ginan build on any Apple M1 hardware.

JohnDonovanGA, 2023-05-30:
stale issue. Closing

closed 2023-05-30 by JohnDonovanGA