GeoscienceAustralia/ginan #11: PPP example does not work well

wangstrong1996, 2022-05-09:
Hi Ginan Team,

Recently, I have tried ginan with the provided example data. However, it seems the PPP results are not very good. Here are the results:

[image]

Should it be like this? 

Best,
Shizhuang Wang

smcclusky, 2022-05-09:
Hi Shizuang,
Can you please post the Ginan config file you used or the example number if you didn't change the config and we will run it here in our development environment server.
Thanks

wangstrong1996, 2022-05-10:
Dear Team,

Thanks for your response. Here is the YAML file we use. We only change the file path in this configuration file.
 
Thank you for your help and for this excellent open-source software.

Best,
Shizhuang

[Uploading ex14_pea_pp_user_gnss_ar_a5.zip…]()

JohnDonovanGA, 2023-05-30:
link is just to this document. closing.

closed 2023-05-30 by JohnDonovanGA