GeoscienceAustralia/ginan #14: Some problem of download the example data

k9891, 2022-07-06:
Hi Ginan Team. Recently, I try to use ginan to some researchs. But, I can't download the example data. The "download_examples.py" returns error of "urllib.error.URLError: <urlopen error [Errno 101] Network is unreachable>", and the link you provided also returns "ERR_CONNECTION_TIMED_OUT" . Is there any other way to download the data? Thanks.

k9891, 2022-07-07:
Hi, bmatv. I try it, but I can't  download any of the top two files manually. I think it's problem of the network. If you are convenient, would you mind sending the example data to my email: kun.c@njnu.edu.cn ? Thanks.

canvaskan, 2022-07-07:
Hi, @k9891 . Thanks for your issue. I encountered same problem with download_example.py, but I managed to dowload data.tar.bz2 and products.tar.bz2 by the link above. I can send you a copy.

bmatv, 2022-07-07:
Hi @k9891 and @canvaskan,
I've published a set of auxiliary products/solutions on github (relevant Ginan v1.4-beta release), this way @k9891's access can be guaranteed. Just clone the repository on your disk and copy over the files to the `ginan/examples`. Here is a link: https://github.com/bmatv/ginan_aux

> Hi, @k9891 . Thanks for your issue. I encountered same problem with download_example.py, but I managed to dowload data.tar.bz2 and products.tar.bz2 by the link above. I can send you a copy.

Thank you for your help, @canvaskan

k9891, 2022-07-07:
> Hi, @k9891 . Thanks for your issue. I encountered same problem with download_example.py, but I managed to dowload data.tar.bz2 and products.tar.bz2 by the link above. I can send you a copy.

Hi, canvaskan. I have downloaded the sample data, thank you very much.

k9891, 2022-07-07:
Hi, Ginan team. Thank you very much!

ck961030
***@***.***

抄送: "幽灵@***@***.******@***.***&gt;;

closed 2022-07-14 by polytopes-design

LJC8848, 2024-05-23:
Hi, @k9891 .I encountered same problem with download_example.py,would you mind sending the example data to my email: 206258902@qq.com ? Thanks.