GeoscienceAustralia/ginan #15: ionmodel part

DoubleString, 2022-07-08:
Hi, ginan team, recently we tried to adopt the coefficient generated function `configure_iono_model_sphcap` provided in `ionosphericalcaps.cpp  ` for ionosphere model, but something went wrong when i called `configure_iono_model_sphcap` and we got a nan output after  `legendre_function` . The while loop can not be broken. The `latc` `lonc` `latw` and `lonw` are set as (23,113,40,40). Is this caused by the code or the inputs? We sincerely look forward to your replys. Further we are confused by the condition code `if((p1 * p1) < 0) at line 367 and 399`, since it will be never fulfilled.

KAHarima, 2022-07-08:
Hi DoubleString,
Unfortunately the B-splines and Spherical Harmonic Caps Ionosphere models are not supported at the moment

KAHarima, 2022-07-08:
If you are interested in testing the function alone, you may want to use radian values for latc lonc latw and lonw

DoubleString, 2022-07-08:
I understand, many thanks!

KAHarima, 2022-07-08:
The expression (p1*p1)<0 is a bug (typo), it should be p1*p2<0. Thank you very much for helping us find it. Unfortunately this model is not maintained, we will add a comment as a warning.

JohnDonovanGA, 2023-05-30:
Issue closed

closed 2023-05-30 by JohnDonovanGA