GeoscienceAustralia/ginan #16: No glonass satellite clock estimation?

jianyaoucar, 2022-07-13:
Dear Geoscience Australia,

I'm Dr. Jian Yao from the U.S. National Center for Atmospheric Research (NCAR/UCAR). We use GNSS softwares to estimate GNSS satellite clocks in the near real time mode, to facilitate the radio occultation. 
We do appreciate very much that Geoscience Australia shares the precious Ginan software. It is pretty easy to use and provides high-quality results, as far as I've tested. One application of Ginan software is to estimate the GPS/GNSS satellite clocks using the global receiver network (i.e., ex17). I've run Ginan for GPS & Galileo satellite clocks and the clock estimation is satisfying. However, I cannot get Ginan to work for Glonass... Please see the attached yaml file for my settings. When I run Ginan, the screen says that all Glonass satellites "failed broadcast predictions"... I've checked the broadcast navigation file brdm1990.19p and it looks like standard. I do notice that the number of fields for Glonass in brdm1990.19p is fewer than that for GPS/Galileo. Is this causing a file-reading issue in Ginan?      I know that your team must be busy with many things. If you can point out which part of the source code that I need to revise, that would be very helpful.

Thank you very much again for sharing Ginan and helping me!

Sincerely,

Jia
[ex17_pea_pp_netw_gnss_ar.yaml.txt](https://github.com/GeoscienceAustralia/ginan/files/9106471/ex17_pea_pp_netw_gnss_ar.yaml.txt)
n

polytopes-design, 2022-07-14:
Hello Jian,
Thanks for your feedback, we hope our software can be of use.
Glonass is not yet officially supported as it has not undergone the amount of testing that other systems have, however I think that your immediate problem can be resolved with a small bugfix for a typo in `src/cpp/rtklib/rinex.cpp`
Toward the bottom of the file, the function `readrnxnav` should read:
```
switch (type)
{
	case 1 : nav.gephMap[geph.Sat][geph.toe]	= geph;	break;
	case 2 : nav.sephMap[seph.Sat][seph.tof]	= seph;	break;
	default: nav.ephMap [eph.Sat] [eph.toe]		= eph;	break;
}
```
by changing case 1 to use `geph.toe` rather than `eph.toe`

We will fix this and many other small bugs, as well as many improvements in the upcoming release.
We hope that this helps.

jianyaoucar, 2022-07-14:
Dear polytopes-design,

Thank you so much for the timely reply! I'll make this change and check the Glonass result.
I do appreciate that the Ginan team is fixing bugs and making many improvements for the future release. Looking forward to it.

Sincerely,

Jian

polytopes-design, 2022-07-15:
One more thing - the `Bias not found in prange for Rxx` warnings you are likely to see are due to a typo in the yaml file - the BIA file should be placed in the config named `bsxfiles:`, not `dcbfiles:`
`bsxfiles: [ TUG0R03FIN_20191990000_01D_01D_OSB.BIA ]`

We will amend the example yaml file to avoid this misunderstanding too.

jianyaoucar, 2022-07-18:
Dear polytopes-design,

This is great to know. Thank you very much for the reminder :)

Sincerely,

Jian

JohnDonovanGA, 2023-05-30:
Issue seems finished. Closing

closed 2023-05-30 by JohnDonovanGA