GeoscienceAustralia/ginan #22: /data cannot be built

hsergi, 2022-11-01:
I did not find a helpdesk contact, so I'm trying here. Sorry if this is not the best way.

I tried installing Ginan with Docker. My issue is that I cannot create the folder /data because I have a corporate machine and would need to ask for permissions and explain why, etc. Is there an alternative to /data? For instance, can /data be anywhere else? If so, how should I run docker? Thanks

umma-zannat, 2022-11-01:
Hi @hsergi,
I do not think you need to create the `/data` folder on your corporate machine, you can use any folder that you like (pick one for which permission is not an issue). But it is convenient to have your folder mount to `/data` _inside_ the docker container. See https://www.freecodecamp.org/news/docker-mount-volume-guide-how-to-mount-a-local-directory/ for clarifications.
So, let's say you would like your data to be in `/home/hsergi/work/ginan-test` or something. You can run your docker image as follows:
```
docker run -it -v /home/hsergi/work/ginan-test:/data gnssanalysis/ginan:v1.5.1 bash
```
Hope this helps. Please let us know how it went.

hsergi, 2022-11-01:
Hello @umma-zannat. Thank you for the tip. I followed it. This is what I got (I had created the conda environment yesterday):

docker run -it -v ~/Documents/GitHub/ginan_data/:/data gnssanalysis/ginan:v1.5.1 bash

And it automatically did:
(gn37) [Ginan] root@0b927b29d80f:/ginan# ls
CHANGELOG.md  ISSUES.md   README.md  aws  bitbucket-pipelines.yml  examples  scripts
Docs	      LICENSE.md  archived   bin  docker		   lib	     src

I have the same structure as the one from GitHub bu with a folder called docker, which has:

Dockerfile  README.md  run-aux.sh  run-tests-long.sh  run-tests-pea.sh	run-tests-pod.sh  tags

Can you tell me if this looks okay? If so, I'll read the docs again to see what's next. Very many thanks.

umma-zannat, 2022-11-01:
I think it looks great. :smile: 
Do you mind if we close the issue now?

hsergi, 2022-11-01:
Absolutely. Thank you so much!

closed 2022-12-13 by hsergi