GeoscienceAustralia/ginan #27: ./README does not install any dependencies

hsergi, 2023-01-10:
I have cloned Ginan in a machine. It's already executable (so the step in the installation that says to make it executable seems outdated):

-rwxr-xr-x 1 srhildeb gipsy 20977 Jan 10 01:38 README.md

When I type in the terminal: ./README.md, nothing at all happens:

[srhildeb@pp1oc ~/ginan]$ ./README.md
[srhildeb@pp1oc ~/ginan]$
[srhildeb@pp1oc ~/ginan]$ ls
archived      docker  examples   LICENSE.md  scripts
CHANGELOG.md  Docs    ISSUES.md  README.md   src

The OS in my machine is: 

NAME="Scientific Linux"
VERSION="7.9 (Nitrogen)"
ID="scientific"
ID_LIKE="rhel centos fedora"

Does the installation from the README.md file only work with Ubuntu?

whatnick, 2023-01-10:
The Ubuntu (apt) and Fedora (yum/dnf) install systems are fundamentally
different. The scientific Linux seems to be a Fedora derivative so under
the yum package management system. You may get some traction by installing
apt equivalent for the Fedora ecosystem.

https://docs.fedoraproject.org/en-US/quick-docs/dnf-vs-apt/

hsergi, 2023-01-12:
P.S. I'll try the source installtion with dnf next week after we upgrade the OS to Red Hat 8.

hsergi, 2023-01-19:
Update: We have installed RHEL8 and dependencies. We are looking at an issue with boost. I'll report soon if we suceed and we can close this issue. Thanks

hsergi, 2023-01-24:
Fine with OpenBLAS. Flags needed in our installation (RHEL8) copied below for cognizance.

I need to test some examples and make sure it runs well. I'll give a final update soon.

cmake -DOpenBLAS_INCLUDE_DIR:PATH=/usr/depot/redhatel8_64/openblas/0.3.21/include  -DOpenBLAS_LIB:FILEPATH=/usr/depot/redhatel8_64/openblas/0.3.21/lib -DBLAS_blas_LIBRARY:FILEPATH=/usr/depot/redhatel8_64/lapack/3.5.0/lib/libblas.so -DLAPACK_LIBRARIES:FILEPATH=/usr/depot/redhatel8_64/lapack/3.5.0/lib/liblapack.so -DNETCDF_LIBRARIES_CXX:FILEPATH=/usr/depot/redhatel8_64/netcdf-c++/4.3.0_ABI0/lib -DNETCDF_INCLUDES_CXX:PATH=/usr/depot/redhatel8_64/netcdf-c++/4.3.0_ABI0/include -DNETCDF_INCLUDES:FILEPATH=/usr/depot/redhatel8_64/netcdf/4.6.2_ABI0/include -DNETCDF_LIBRARIES:FILEPATH=/usr/depot/redhatel8_64/netcdf/4.6.2_ABI0/lib ../

hsergi, 2023-01-26:
Since ex11 is advertised, MongoDB is required, so I would remind the user about it. Everything else seems fine. P.S. the tar zip files are easy to download in a machine with good internet connection. In cases where the connection is not good enough (unfortunately, the case of my institutional secure servers), I download it to my laptop and upload the file via sftp. Then, I had to unzip it and re-run ``python3 scripts/download_examples.py``

hsergi, 2023-01-30:
Okay. MongoDB is optional, and I set to ``false`` the option in ex11. It run well.

If someone uses RHEL, please see above for tips about the installation of external packages.

closed 2023-01-30 by hsergi