GeoscienceAustralia/ginan #33: Estimation of ionosphere and DCB

chuanfengzhu, 2023-04-09:
Dear Ginan term
I am recently working on the estimation of ionosphere and DCB. While separating the receiver DCB and satellite DCB, I didn't find any information in the Ginan program about how the reference constraint is handled. I would appreciate it if you could explain this issue to me. Thank you very much, and I look forward to your reply.

KAHarima, 2023-04-11:
Hi chuangfengzhu
Thank you for your interest on Ionosphere/DCB estimation in Ginan. Short answer to your question is that Ginan fixes the DCB for a specific station to zero. This pivot station is set in one of two ways:
1. The user can set the "reference" station by designating it on **processing_options -> gnss_general -> pivot_station**  
2. If no station is designated (or the designated station does not have data) the station is selected (in alphabetic order) from those stations with the maximum number of visible satellites.
In the event of an outage on the pivot station, a new station will be selected as pivot and the DCBs will be reset

chuanfengzhu, 2023-04-11:
> Hi chuangfengzhu Thank you for your interest on Ionosphere/DCB estimation in Ginan. Short answer to your question is that Ginan fixes the DCB for a specific station to zero. This _pivot station_ is set in one of two ways:
> 
> 1. The user can set the _pivot station_ by designating it on **processing_options -> gnss_general -> pivot_station**
> 2. If no station is designated (or the designated station does not have data) the _pivot station_ is selected (in alphabetic order) from those stations with the maximum number of visible satellites.
>    In the event of an outage on the _pivot station_, a new station will be selected as pivot and the DCBs will be reset
Dear KAHarima
Thank you for your patient explanation. I changed the constraint of a specific station's DCB to zero to the constraint that the sum of all satellite DCBs is zero. The resulting satellite DCBs are similar in value to the DCB products of CAS/DLR, but the calculated station DCBs are somewhat unusual in value. I am not sure if my constraint method is incorrect. Could you give me some constructive suggestions for modification? Here are some of the satellite and station DCBs that I obtained using the satellite zero-mean condition:
[image]