GeoscienceAustralia/ginan #45: When I build ginan on Mac, error: ld: library not found for -licudata occurred.

WUJing1129, 2023-07-24:
When I build ginan on Mac, error: ld: library not found for -licudata occurred.

smcclusky, 2023-07-24:
Hi WUJing1129,
This is likely a problem with your boost installation. Can you please tell me what version of Boost you have installed and are using? Ginan on the Mac supports only up to v1.73

WUJing1129, 2023-07-26:
Hi Simon,
Thanks for your information. I changed my boost back to v1.73, and this problem has been fixed.
Thanks for your team's effort to open-source such powerful software.
WU Jing

[https://www.polyu.edu.hk/emaildisclaimer/PolyU_Email_Signature.jpg]

Disclaimer:

This message (including any attachments) contains confidential information intended for a specific individual and purpose. If you are not the intended recipient, you should delete this message and notify the sender and The Hong Kong Polytechnic University (the University) immediately. Any disclosure, copying, or distribution of this message, or the taking of any action based on it, is strictly prohibited and may be unlawful.

The University specifically denies any responsibility for the accuracy or quality of information obtained through University E-mail Facilities. Any views and opinions expressed are only those of the author(s) and do not necessarily represent those of the University and the University accepts no liability whatsoever for any losses or damages incurred or caused to any party as a result of the use of such information.

closed 2023-07-26 by seballgeyer