GeoscienceAustralia/ginan #47: Real-time stream connect problem: wrong version number

whu-dyf, 2023-07-27:
Dear Ginan Team,

I am writing to report some errors that I encountered while attempting to record streams using Ginan. Initially, I suspected that my stream configuration might be incorrect, but it works well with rtklib and BNC.

Furthermore, I tried to use curl to test the streams, as outlined in your documentation. While the command line in your example (as shown in Figure 2) worked well, my own stream configuration did not. I would greatly appreciate it if you could provide me with more information regarding the stream format. Specifically, I am unsure whether the port (such as: 2101) should be included in input stream format or not.

Thank you in advance for your assistance.
[image]
[image]

whu-dyf, 2023-07-28:
Even if I run record_streams.yaml in the exampleConfigs, it still can not work. I just put my username and password in the <USER> and <PASS>.
[image]
[image]

closed 2023-07-31 by whu-dyf

WhuGcLC, 2024-06-11:
Hello, I have the same questions as you.Could you tell me how to solve this problem?