GeoscienceAustralia/ginan #51: GINAN Software Processing Time Help Request

YaoChunKo, 2023-09-18:
I have been using GINAN version 1.5 and the latest version 2.1 for my research. I encountered a significant difference in processing time with GINAN version 2.1 when compared to GINAN version 1.5, and I have some questions regarding this.

I initially ran the ex204.yaml example with GINAN version 2.1, where I made some modifications to obtain the output of GLONASS clock products. However, I observed that the processing time for this task was around 3 days. To validate this, I downloaded GINAN version 1.5 and performed a similar test using the ex17_pea_pp_netw_gnss_ar.yaml example. I ensured that the settings, such as epoch control, GLONASS clock computation, and data frequency (30 sec RINEX), were aligned with those in the ex204.yaml for GINAN version 2.1. Surprisingly, the processing time for the ex17.yaml test in version 1.5 was significantly faster, taking only about 57 minutes.

I conducted both tests on the same computer, eliminating any hardware variations. Additionally, the setting of the subfiles used in the ex204.yaml of GINAN version 2.1 are similar to those in ex17.yaml. My goal is to understand the factors contributing to the longer processing time in version 2.1 and whether there are any settings or optimizations that can be applied to speed up the processing in GINAN version 2.1.

smcclusky, 2023-09-19:
Hi Yao-Chun,
Ginan v1.5 uses an undifferenced (UD) ionosphere free linear combination (IFLC) of observations, Ginan v2 uses a undifferenced (UD) uncombined (UC) approach. The Ginan v2 performance hit is expected when you use observations derived from many codes (defined in the **code_priorities:** config option) simultaneously. Using many observation codes leads to a large increase in the number of bias and ambiguity States in the State estimation filter. You need to keep in mind that epoch by epoch processing time goes up by approximately the cube (power 3) of the size of the Kalman Filter State Vector length. So if your State vector using Ginan v2 is 4x longer than in v1.5 (which is easily possible if you include many codes) then your processing time will go up by a factor 64 (ie 4^3). So you can see how your 1 hour v1.5 processing time can go to 64 hours using v2 very easily if you are not careful. 
The good news is there is a way to configure Ginan v2 to essentially reproduce the original Ginan v1 two frequency ionosphere free combination solution for estimation of Satellite Clocks (and other States) as I think you are attempting to do? This is essentially what we use operationally at GA to produce IGS RAPID like products. This v2 solution slightly slower (factor 1.5) than the original Ginan v1.5 solution, but the slowdown should not be to the extent you saw in your example. I am just testing the GLO modification to that configuration at the moment, once I am happy it is working well I will post it for you to try. Apologies for the delay getting back to you on this.

YaoChunKo, 2023-09-21:
Hi Simon,

Thank you for providing all the details! Your explanation regarding the computation speed is clear and reasonable to us.

Yes, factor of 1.5 slower than Ginan 1.5 is acceptable. Please take your time on the modification of Ginan v2.1 and let us know after the modification is done.

closed 2024-03-12 by seballgeyer