GeoscienceAustralia/ginan #52: Supported Platforms v. Build/Installation Scripts

liam-curran, 2023-10-04:
Not really a bug, nor necessarily wrong, but maybe an inconsistency on the main page & build/installation Scripts:

There are scripts to install dependencies for Fedora 38, but it is not noted as an Installation from source / Supported Platforms. Is this missed cross reference or Fedora 38 isn't strictly supported?

jiargei, 2023-10-23:
Dear GA!

@liam-curran : there is an error in the bash script scripts/installation/fedora38.sh, it seems GA used a Dockerfile as blue print.

It will fail in line 92: replace "RUN" with "$sudo_cmd" or just "sudo" in case you're not root. I changed it and ran it under Rocky Linux 8 and it worked!

It would be nice if you would provide an ansible playbook for preparing a server for ginan.
It is just an idea..

kind regards

seballgeyer, 2023-10-31:
Thanks for this, the issue is fixed in the `develop_weekly` branch.

closed 2023-10-31 by seballgeyer

jiargei, 2023-11-03:
I wrote an ansible role to install GINAN on an Rocky Linux 8 (RHEL 8). Should I provide it to you anyhow?