GeoscienceAustralia/ginan #55: Linking against boost fails

meerfrau, 2023-10-31:
After building to 100 % (gcc 13.2, boost 1.83) the linking fails due to undefined references:
```
/usr/bin/ld: ./ginan/lib/libginan_core.a(acsConfig.cpp.o): in function `boost::log::v2s_mt_posix::aux::record_pump<boost::log::v2s_mt_posix::sources::severity_logger_mt<boost::log::v2s_mt_posix::trivial::severity_level> >::auto_release::~auto_release()':
/usr/include/boost/log/sources/record_ostream.hpp:493:(.text.unlikely+0xe9d): undefined reference to `boost::log::v2s_mt_posix::aux::stream_provider<char>::release_compound(boost::log::v2s_mt_posix::aux::stream_provider<char>::stream_compound*)'
/usr/bin/ld: ./ginan/lib/libginan_core.a(acsConfig.cpp.o): in function `boost::log::v2s_mt_posix::record::reset()':
/usr/include/boost/log/core/record.hpp:157:(.text.unlikely+0xeb0): undefined reference to `boost::log::v2s_mt_posix::record_view::public_data::destroy(boost::log::v2s_mt_posix::record_view::public_data const*)'

[..]

/usr/bin/ld: ./ginan/lib/libginan_core.a(acsConfig.cpp.o):(.data.rel.ro._ZTVN5boost15program_options11typed_valueISt6vectorINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESaIS8_EEcEE[_ZTVN5boost15program_options11typed_valueISt6vectorINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESaIS8_EEcEE]+0x38): undefined reference to `boost::program_options::value_semantic_codecvt_helper<char>::parse(boost::any&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, bool) const'
/usr/bin/ld: ./ginan/lib/libginan_core.a(acsConfig.cpp.o):(.data.rel.local.DW.ref._ZTIN5boost10filesystem16filesystem_errorE[DW.ref._ZTIN5boost10filesystem16filesystem_errorE]+0x0): undefined reference to `typeinfo for boost::filesystem::filesystem_error'
/usr/bin/ld: warning: creating DT_TEXTREL in a PIE
```

Are the special requirements regarding C++ mode or max. Boost version?

seballgeyer, 2023-10-31:
Hello, 

We haven't tested Ginan with GCC13, we will have a look at it. 

Could you please provide more information on the operating system you are using? 

Thanks,

jiargei, 2023-11-03:
I also had problems with Boost 1.78. I built Boost with 1.73 and it worked! (Rocky Linux 8.5)

seballgeyer, 2023-11-03:
Hello

could you provide more information on how both boost libraries were installed so we can try to reproduce the issue?

Thanks

jiargei, 2023-11-04:
In my case (Rocky Linux 8) I had problems with the default boost version
1.78. The CMake could find it but during the make process it failed because
it had no link to e.g. libboost_timer.a. I checked RL8 folder structure and
there was no such file, but there was a libboost_timer.so.1.78.0 file. I
checked the CMakeCache.txt and all vthe boost links had the same issue. By
building libboost (1.73) this issue was resolved. I haven't checked it with
a newer version.

Kind regards

seballgeyer, 2023-11-04:
From what you are saying, one of the issues is that GINAN is looking for static libraries (*.a). This is in the CMakelist.txt (line 174 ). You can try to comment on this line.

jiargei, 2023-11-08:
It works now. One thing.. At least in Rocky Linux 8 i can compile it with boost 1.73 and 1.74. It fails with 1.76. One more issue is resolved with Pullrequest #58 (static_pointer_cast was not found).

jiargei, 2023-11-28:
I just pulled the latest commit from develop-weekly branch. The pull request #58 would fix import issue on algebraTrace.cpp.

closed 2024-05-07 by ronaldmaj