GeoscienceAustralia/ginan #57: PEA crashes on editing main yaml config - "inputs:root_directory" issue

jiargei, 2023-11-05:
**Describe the bug**
Editing and Saving results in Reloading the yaml config. I think that's how it should be. While loading the ATX the reload fails. It seems that with the reloading the inputs:root_directory parameter will be attended. When I set inputs:root_directory to "." and set all the files to "./products/..." one can see that with every reload the "./" will be attended to loading the ATX file. I edited one parameter in the main yaml (the one that loads all the other ones) and I could see after 4 time editing that the path to ATX is now: "././././igs20.atx"
This occurs at least on Real-Time Network processing.

**To Reproduce**
Steps to reproduce the behavior:
1. Define inputs:root_directory: products/
2. Start pea -y config.yaml
3. Edit and save something in config.yaml
4. PEA will reload and failing on ATX

```
Checking configuration file asbru.yaml
Loading configuration from file input_metadata.yaml
Loading configuration from file input_streams_rover.yaml
Loading configuration from file processing_network.yaml
Loading configuration from file debug_trace.yaml
Loading configuration from file config.yaml
Error: Invalid  file products/products/igs20.atx
Message met fatal_message_level condition for exit.
Exiting...

Segmentation fault (core dumped)

```

**Expected behavior**
PEA stops

**Configuration files**
Yaml files assist with with confirming the exact processing options of the software. Default yaml from exampleConfigs

**Desktop (please complete the following information):**
- OS: Ubuntu 22.04
- Version of Ginan: v2.1.0
- Library used (Boost 1.74, Eigen 3.4)

**Additional context**
Add any other context about the problem here.

seballgeyer, 2023-11-05:
Thanks for reporting this issue. 

Could you try to build and test the code present on the `develop_weekly` branch (our weekly development snapshot)?

Let us know if the issue is resolved in the `develop_weekly` branch. If not, we will investigate further.

Thank you

jiargei, 2023-11-06:
Hej!
Yes, it seems better now! The pea output looks also cleaner now!
Would you mind adding the Pull-Reuest #54 (https://github.com/GeoscienceAustralia/ginan/pull/54) to your weekly commit? It just corrects the Network*TRACE output so that the Section "+SITE/DATA" will be closed with "-SITE/DATA". right now it is "-Site Data" which looks different than the beginning of the section!
kind regards!

jiargei, 2023-11-07:
Tested:Ubuntu 22.04, Eigen 3.4.0, Boost 1.74.0
ToDo: RockyLinux 8 (RHEL 8)

closed 2023-11-08 by jiargei