GeoscienceAustralia/ginan #61: fit_sp3_pseudoobs

mgrutten, 2023-12-28:
Hi Team!

I'm using the `release-v3.0-alpha` branch, so I understand if it's a work in progress.

The reason I'm looking at that branch is the `fit_sp3_pseudoobs.yaml` example, which I think generates data very similar to what I'm after.  I would like to take an existing (set of) sp3 files, fit orbits and precisely predict those orbits a day into the future.

I can run the example (using the docker image) and generate sp3 files, but I don't understand the output.  The ephemeris that is generated (3 sp3 files) is over the same time-period as the input files and doesn't match the input at all?  I've used the `compare_orbits.yaml` example to compare them and the distances are in the 1000s of km.

The only modification to the `fit_sp3_pseudoobs.yaml` file was to comment out the "mongo" section, as I'm not using the visualisation.

I'm sure it's a misunderstanding, since I'm very new to ginan.  Could you please explain the output?

polytopes-design, 2023-12-28:
Hello mgrutten.
To address part of your query - to predict one day into the future the `end_epoch` configuration can be set to the end of the desired propagation. Use `pea -Y 4` to see all yaml config parameters and their locations within the yaml file, but that parameter will be in the `epoch_control` section.
You will likely also need to set `require_obs` to false, so that the prediction will progress despite the sp3 files having exhausted their data.

As for the bad fits - it appears that the fit_sp3_pseudoobs.yaml is set up to `output_inertial` values to the sp3, so the results are in a different reference frame - remove this config and the results will be in an ecef frame as per the input files.

Thanks for your feedback, our input helps us improve the software.

mgrutten, 2023-12-29:
Hi @polytopes-design 

Thanks for the very quick and very helpful reply!
I've made both of the changes you suggested (`epoch_control` and `output_inertial`) and the output is exactly what I was hoping to see.  Comparing the predictions against the FIN sp3 shows they align very closely (for my purposes).

It's very likely that I will have a few more questions.  Is there a ginan forum, or is this the most appropriate place to ask?

Mark.

closed 2024-03-12 by seballgeyer