GeoscienceAustralia/ginan #64: Has anyone successfully compiled this version 3.0 in ubuntu18?

tachien, 2024-02-13:
The 3.0 version of the software seems to need higher requirements for C++ compilers, and I couldn't successfully compile this software in ubuntu18. Error shows that no matching function for call to 'std::allocator<char>::allocate(std::size_t&,char*)'. Has anyone else had this problem?

seballgeyer, 2024-02-14:
Hello

It has been a while since we haven't used Ubuntu18. 

In order to help us, could you please provide use with details on the compiler, and library version that are installed on your system:
- Version of the compiler:
- Version of the boost library:
- Version of the mongo-cpp driver:

All those information should be on your terminal output when running `cmake ../.` 

Thank you

tachien, 2024-02-14:
Thanks for your kind reply. I compile the 3.0 version in ubuntu 22.03 TLS. However, the same question. My compiler is gcc/g++ in 11.4. My boost library is in version of 1.73. My mongo-cxx-driver is in the version of r3.8.0. My mongo-c-driver is in the version of 1.24.1.

seballgeyer, 2024-02-14:
For ubuntu 22.04 LTS, we have instruction in the `ginan/scripts/installation` directory. The file is called ubuntu22.sh. 

You are using boost 1.73. This version has problems with gcc/g++ 11 or higher. We suggest using boost 1.74 or higher with gcc 11 or higher.

tachien, 2024-02-14:
Thx! The problem has been fixed!