GeoscienceAustralia/ginan #68: AArch64/Arm64

liam-curran, 2024-03-11:
Is there any reason why AArch64/Arm64 docker containers aren't available (or at least don't seem to obvious on docker)?

Thinking for recent Apple silicon Macs or Raspberry Pi etc. thanks

ronaldmaj, 2024-03-13:
Hi Liam,

Thanks for your question.
At the moment we're not developing Ginan with any AArch64/Arm64 hardware in mind, so we don't have a Dockerfile or regular build process for that purpose.

However, in a future update of the repo we will be including the Dockerfile for creating the "ginan-env" image - used in turn to build our main Ginan Docker images. You should be able to edit the "ginan-env" Dockerfile for the purpose of creating a AArch64/Arm64 Docker image - if you do so, you are welcome to create a Pull Request from a Fork of Ginan, to contribute this back.

Let us know how you go and if you have any further questions or concerns, please don't hesitate to reach out :)

liam-curran, 2024-03-13:
Thanks Ronald, I'm not really in the dev space, the question more relates to future hardware considerations. Maybe a bit of a hope after seeing a past presentation about "Ginan in a box" on a RPi. I'll  look more into ginan-env. Happy to look at testing for this architecture if it's considered valuable for the project and not too niche. Thanks again.

ronaldmaj, 2024-03-15:
Hi Liam,
Yeah definitely, testing on Raspberry Pis (and other platforms) is of great value for the project as it allows us to spot and correct issues found in the software. 
On the Ginan-in-a-box project you referenced, we actually worked closely with FrontierSI on doing that testing - they were able to take our base Dockerfile and create a bespoke image for their Raspberry Pi. However, we have not operationalised any process to build that regularly - something that we may do so in the future but not in the near term.