GeoscienceAustralia/ginan #74: Use of GinanEDA

gnss-freshman, 2024-03-20:
**Describe the bug**
ginan/scripts/GinanEDA/requirements.txt, the libraries in it have also been installed successfully
I followed https://geoscienceaustralia.github.io/ginan/page.html?c=on&p=scripts.index that tutorial using ginaneda and have not been able to open http://127.0.0.1:5000.
[image]
Also I thought I'd ask how ginan/scripts/backup_old/compare_pod_rms.py should be used, I didn't find the pod.out pod.rms as required in the code, could you please tell me how they are formatted?

seballgeyer, 2024-03-20:
Hello, 

The output seems to be correct. 
For completeness, are you running the EDA on your local machine or a remote one? It might be an issue of forwarding the ports if you are using a remote host. 

As for the other question, all scripts in backup_old, relates to old versions of Ginan, you will not be able to use them. The output and architecture of the software has changed.

Best regards,
Sebastien

gnss-freshman, 2024-03-20:
Okay, thanks, I'm running the EDA remotely, and I was wondering what other programs besides EDA are available to evaluate the accuracy of the orbiting and clock counting.

seballgeyer, 2024-03-20:
Hello,

It seems like the EDA (Error Detection and Analysis) is running on a remote host, which makes it inaccessible directly from your localhost. To access it on your localhost, you'll need to forward the port 5000 of your remote host to your localhost. There are several ways to do this, but one common method is using SSH port forwarding.

You can forward the port using the following command:
```bash
ssh -fN -L 5000:localhost:5000 <remoteusername>@<remoteip>
```

This command forwards port 5000 from the remote host to port 5000 on your localhost. Replace <remoteusername> with your username on the remote host and <remoteip> with the IP address of the remote host.

Regarding your second question, for post-processing and analyzing the output, additional programs are available. We have organized them into a new repository. You can find them at  https://github.com/GeoscienceAustralia/gnssanalysis .

If you need further assistance or clarification, feel free to ask.

Best regards,
Sebastien

gnss-freshman, 2024-03-20:
Thank you very much for your answer, I am currently running GINANEDA successfully and I will follow up on the second accuracy rating!
[image]