GeoscienceAustralia/ginan #82: A memory leak occurs during real-time precise point positioning

2533672343, 2024-03-29:
**Describe the bug**
Running multi-station real-time precision point positioning has the problem of memory leak.

**To Reproduce**
Steps to reproduce the behavior:
1. ./../bin --config ex15_pea_rt_user_gnss_ar.yaml
2. Enter the 'top' command on the terminal, and you can see that the VIRT value continues to increase.

**Expected behavior**
The amount of memory used during program execution does not always increase over time.

**Configuration files**
ex15_pea_rt_user_gnss_ar.yaml, and the ambiguity_resolution is off, delete_old_ephemerides is true.

**Desktop (please complete the following information):**
- OS: Linux 20.04
- Version of Ginan:1.5.4
- Library used (Boost, Eigen, ...)

seballgeyer, 2024-04-14:
Hello 

It seems you are experiencing a memory leak issue with Ginan version 1.5.4. Memory leaks can be problematic as they cause the application to consume more and more memory over time, which can eventually lead to performance issues or system crashes.

The first step in addressing this issue is to ensure that you are using the latest version of the software. As you mentioned, you are using Ginan 1.5.4 which is outdated, and updating to the latest version, which is version 3.0.0, could resolve the memory leak problem. The new version includes many improvements such as performance enhancements and bug fixes, which might include a fix for the memory leak issue.

Here’s a general outline of steps you could take to resolve the issue:

1. Obtain the latest version of Ginan from the official repository.

2. Configuration Check: Update your configuration files.

If the problem persists, feel free to contact us.

2533672343, 2024-04-21:
Hello

Thank you for your reply. But the memory leak problem seems to exist in version 3.0.0 as well.

2533672343, 2024-04-21:
Hello

Thanks for your response, and the memory leak problem seems to exist in version 3.0.0 as well.

seballgeyer, 2024-04-23:
Hello,

Could you provide the configuration file you are using? 

Thank you.

2533672343, 2024-04-26:
Hello,
Thank you for your reply.
Memory is still slowly increasing after a day of operation. Is it normal ?
The configuration file(rt_ppp_example.yaml) is as follows:
inputs:

    inputs_root: ./products/

    atx_files:  [ igs20.atx                             ]
    egm_files:  [ tables/EGM2008.gfc                    ]
    igrf_files: [ tables/igrf13coeffs.txt               ]
    erp_files:  [ finals.data.iau2000.txt        ]
    planetary_ephemeris_files: [ tables/DE436.1950.2050 ]

    troposphere:
        gpt2grid_files: [ tables/gpt_25.grd ]
    
    tides:
        ocean_tide_loading_blq_files:   [ tables/OLOAD_GO.BLQ                  ]
        atmos_tide_loading_blq_files:   [ tables/ALOAD_GO.BLQ                  ]
        ocean_pole_tide_loading_files:  [ tables/opoleloadcoefcmcor.txt ]
        ocean_tide_potential_files:     [ tables/fes2014b_Cnm-Snm.dat   ]

    snx_files:  [ IGS0OPSSNX_20233020000_07D_07D_SOL.SNX ]

    gnss_observations:
        gnss_observations_root: "https://<user>:<password>@ntrip.data.gnss.ga.gov.au/"
        rtcm_inputs:
            - ALIC00AUS0
            - MAW100ATA0
            - DARW00AUS0
            - STR200AUS0

    satellite_data:
        satellite_data_root: "https://<user>:<password>@ntrip.data.gnss.ga.gov.au/"
        rtcm_inputs:
            ssr_antenna_offset: APC
            rtcm_inputs:
            - BCEP00BKG0
            - SSRA00BKG0

outputs:
    metadata:
        config_description:    rt_ppp_example

    outputs_root:              ./outputs/<CONFIG>

    trace:
        level:                 2
        output_receivers:      false
        output_network:        false
        receiver_filename:     <CONFIG>_<RECEIVER>_<YYYY><DDD><HH>.TRACE
        network_filename:      <CONFIG>_<RECEIVER>_<YYYY><DDD><HH>.TRACE
        output_residuals:      true
        output_residual_chain: true
        output_config:         true

    gpx:
        output:                true
        filename:              <CONFIG>_<RECEIVER>_<YYYY><DDD><HH>.GPX

satellite_options:

    global:
        models:
            pos:
                enable:             true
                sources:            [ SSR ]
            clock:
                enable:             true
[... 102 lines cut ...]
                sigma:              [0.03]
                process_noise:      [1.0E-6]
            code_bias:
                estimated:          [true] # false
                sigma:              [20]
                process_noise:      [0]
            phase_bias:
                estimated:          [false]
                sigma:              [10]
                process_noise:      [0]

mongo:                   # Mongo is a database used to store results and intermediate values for later analysis and inter-process communication

    enable:              none                     # Enable and connect to mongo database {none,primary,secondary,both}
    primary_uri:         mongodb://localhost:27017   # Location and port of the mongo database to connect to
    primary_database:    <CONFIG>
    output_components:   primary                     # Output components of measurements {none,primary,secondary,both}
    output_states:       primary                     # Output states {none,primary,secondary,both}
    output_measurements: primary                     # Output measurements and their residuals {none,primary,secondary,both}
    output_test_stats:   primary                     # Output test statistics {none,primary,secondary,both}
    delete_history:      primary                     # Drop the collection in the database at the beginning of the run to only show fresh data {none,primary,secondary,both}

debug:

    instrument: true

jiargei, 2024-10-28:
Is that correlated to #119 ?