GeoscienceAustralia/ginan #90: Does Ginan support triple-frequency uncombined PPP and PPP-AR?

Levi-always-rejoice, 2024-06-06:
Dear Ginan team,

  I am wondering if Ginan software could execute triple-frequency uncombined PPP. The parameters are huge and sophisticated. I have tried to use the Ginan software to produce a result of triple-frequency uncombined PPP with ambiguity fixed under the guide of **Config Parameters** but  I failed according to the result. So my question concerning Ginan is as follows.
    (1) Does Ginan support PPP in triple-frequency uncombined mode? If yes, how could I configure the parameters in the YAML file that controls the number of frequency being used in the PPP processing?
    (2) Does Ginan support PPP-AR for triple-frequency uncombined PPP? If yes, how could I configure the parameters in the YAML file that enable PPP-AR for triple-frequency uncombined PPP?
    In addition, I am wondering if you provide a tutorial on debugging Ginan code?

Levi-always-rejoice, 2024-07-12:
Why nobody answer my questions?

smcclusky, 2024-07-17:
Hi Levi-always-rejoice,
I have had a look at the triple frequency example you mention. I have successfully run triple frequency GPS + GAL PPP solutions for a few stations without ambiguity resolution (AR) though I am still working to sort out a few problems with phase bias estimation and AR in tri-freq mode.
So to answer your questions:
1) Yes - Ginan. dose support tri frequency PPP in uncombined mode. Included below is a config file to this reply that should work for you (but not with AR yet). You will need to use the latest Ginan weekly snapshot with this config file.
(https://github.com/GeoscienceAustralia/ginan/tree/develop-weekly)
2) Not yet - Ginan does support AR in dual frequency mode, I am still solving some issues in tri-frequency mode.
3) We do not have a Ginan code debugging tutorial available. If you are interested in starting to develop / modify Ginan code, please send me some details of your area interest I will see if I can connect you directly with the Ginan software engineering team.
Regards
--Simon (simon.mcclusky@ga.gov.au)
-----------------------
```
inputs:

    inputs_root: ./products/

    atx_files:  [ igs20.atx                             ] # required
    igrf_files: [ tables/igrf13coeffs.txt               ]
    erp_files:  [ finals.data.iau2000.txt               ]
    planetary_ephemeris_files: [ tables/DE436.1950.2050 ]

    troposphere:
        gpt2grid_files: [ tables/gpt_25.grd ]

    tides:
        ocean_tide_loading_blq_files:   [ tables/OLOAD_GO.BLQ                  ] # required if ocean loading is applied
        atmos_tide_loading_blq_files:   [ tables/ALOAD_GO.BLQ                  ] # required if atmospheric tide loading is applied
        ocean_pole_tide_loading_files:  [ tables/opoleloadcoefcmcor.txt ] # required if ocean pole tide loading is applied

    snx_files:
    # - "*.SNX" # use a wild card to include all files matching the description in the directory
    - igs_satellite_metadata.snx
    - tables/sat_yaw_bias_rate.snx
    - tables/qzss_yaw_modes.snx
    - tables/bds_yaw_modes.snx
    - IGS1R03SNX_20191950000_07D_07D_CRD.SNX

    satellite_data:
        # nav_files:
        # - "*.rnx"
        clk_files:
        # - "*.CLK"
        - IGS2R03FIN_20191990000_01D_30S_CLK.CLK
        bsx_files:
        # - "*.BIA"
        #- IGS2R03FIN_20191990000_01D_01D_OSB.BIA
        - TUG0R03FIN_20191990000_01D_01D_OSB.BIA
        sp3_files:
        # - "*.SP3"
        - IGS2R03FIN_20191990000_01D_05M_ORB.SP3

    gnss_observations:
        gnss_observations_root: ../data/
        rnx_inputs:
        # - "*.rnx"
        - ALIC00AUS_R_20191990000_01D_30S_MO.rnx
        - DARW00AUS_R_20191990000_01D_30S_MO.rnx
        - HOB200AUS_R_20191990000_01D_30S_MO.rnx
        #- KARR00AUS_R_20191990000_01D_30S_MO.rnx
        #- TOW200AUS_R_20191990000_01D_30S_MO.rnx
        #- "M*.rnx"

outputs:

    metadata:
        config_description:    ppp_example_gps+gal_L1L2L5

    outputs_root:              ./outputs/<CONFIG>

    gpx:
        output:                true
[... 246 lines cut ...]
                    sigma:              [0.5]
                    process_noise:      [0.0002]
            L5X:
                phase_bias:
                    estimated:          [true]
                    sigma:              [0.5]
                    process_noise:      [0.0002]

mongo:                   # Mongo is a database used to store results and intermediate values for later analysis and inter-process communication
    enable:              primary                     # Enable and connect to mongo database {none,primary,secondary,both}
    primary_uri:         mongodb://localhost:27017   # Location and port of the mongo database to connect to
    primary_database:    <CONFIG>
    primary_suffix:      ""                          # Suffix to append to database elements to make distinctions between runs for comparison
    # secondary_database:  <CONFIG>
    # secondary_suffix:    ""                          # Suffix to append to database elements to make distinctions between runs for comparison
    # secondary_uri:       mongodb://localhost:27017   # Location and port of the mongo database to connect to
    # output_config:       NONE                        # Output config {none,primary,secondary,both}
    output_components:   primary                     # Output components of measurements {none,primary,secondary,both}
    output_states:       primary                     # Output states {none,primary,secondary,both}
    output_measurements: primary                     # Output measurements and their residuals {none,primary,secondary,both}
    output_test_stats:   primary                     # Output test statistics {none,primary,secondary,both}
    delete_history:      primary                     # Drop the collection in the database at the beginning of the run to only show fresh data {none,primary,secondary,both}
    output_trace:        primary
    #output_cumulative:   primary
```

Levi-always-rejoice, 2024-07-17:
Thank you Simon McClusky. Thank you for your detailed answer.

closed 2024-07-25 by ronaldmaj