GeoscienceAustralia/ginan #92: Failed to run pea by using ppp_example.yaml on ubuntu22.04

wsaddasw1989, 2024-06-13:
Dear ginan team,
I install the ginan on ubuntu22.04 according to the README.md file and using the ubuntu22.sh script. Everything is good during the installing and complining process, but I got "NAN foud. Exiting ..." ERROR in the first epoch process while runing "./pea --config ppp_example.yaml".  All the files and yaml are default ones and the inputdata is well prepared. There are the  same problem while runing on CentOS8.  Here is the cmake info and run info on ubuntu22.04.  Is there any problem? Thank you.
`-- The C compiler identification is GNU 11.4.0
-- The CXX compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Setting ccache            off
-- Setting optimisation      on
-- Setting parallelisation   on
-- Found OpenMP_C: -fopenmp (found version "4.5") 
-- Found OpenMP_CXX: -fopenmp (found version "4.5") 
-- Found OpenMP: TRUE (found version "4.5")  
-- Setting docs              on
-- Found Doxygen: /usr/bin/doxygen (found version "1.9.1") found components: doxygen dot 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Found OpenSSL: /usr/lib/x86_64-linux-gnu/libcrypto.so (found version "3.0.2")  
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake (found suitable version "1.74.0", minimum required is "1.73.0") found components: log log_setup date_time system thread program_options serialization timer stacktrace_addr2line stacktrace_backtrace 
-- Found NetCDF: /usr/lib/x86_64-linux-gnu/libnetcdf_c++4.so;/usr/lib/x86_64-linux-gnu/libnetcdf.so  
-- Could NOT find BLAS (missing: BLAS_LIBRARIES) 
-- Looking for sgemm_
-- Looking for sgemm_ - not found
-- Looking for sgemm_
-- Looking for sgemm_ - found
-- Found BLAS: /usr/lib/x86_64-linux-gnu/libopenblas.so  
-- Looking for cheev_
-- Looking for cheev_ - found
-- Found LAPACK: /usr/lib/x86_64-linux-gnu/libopenblas.so;-lm;-ldl  
-- Found LAPACK and BLAS
-- Found YAML library:       /usr/lib/x86_64-linux-gnu/libyaml-cpp.so
-- Found NetCDF library:     /usr/lib/x86_64-linux-gnu/libnetcdf_c++4.so
-- Found Eigen version:      3.4.0
-- Found Boost version:      1.74.0
-- Found Mongocxx version:   3.6.7
-- Found C++ compiler:       GNU 11.4.0
-- Found Git: /usr/bin/git (found version "2.34.1") 
-- Git branch tag:           untagged-0f26327-dirty
-- Git branch:               main
-- CMAKE_CURRENT_BINARY_DIR: /ginan/src/build
-- CMAKE_CURRENT_SOURCE_DIR: /ginan/src
-- CMAKE_CXX_FLAGS:          -std=c++2a -fpermissive -Wall -Wno-write-strings -Wno-deprecated-declarations -Wno-format-overflow -Wno-narrowing -Wno-unused-but-set-variable -Wno-sign-compare -Wno-stringop-truncation -Wno-unused-variable -Wno-switch -Wno-dangling-else -Wno-misleading-indentation -Wno-format-truncation -Wno-extern-c-compat -Wno-format-zero-length -Wno-array-bounds -Wno-restrict -O3 -fopenmp -pthread
-- Configuring done
-- Generating done
-- Build files have been written to: /ginan/src/build
`

`
PEA starting... (main untagged-0f26327-dirty from Tue Apr 30 04:29:34 2024)

Checking configuration file ppp_example.yaml
Loading configuration from file ppp_example.yaml

===============================
Configuration...
===============================
Inputs:
	snx_files:                       ./products/igs_satellite_metadata.snx ./products/tables/sat_yaw_bias_rate.snx ./products/tables/qzss_yaw_modes.snx ./products/tables/bds_yaw_modes.snx ./products/IGS1R03SNX_20191950000_07D_07D_CRD.SNX 
	atx_files:                       ./products/igs20.atx 
	clk_files:                       ./products/IGS2R03FIN_20191990000_01D_30S_CLK.CLK 
[... 143 lines cut ...]
     2.23874
           0
           0
           0
           0
     22.5539
......

xp:
         -nan
         -nan
......

R :
......

v :
  -2.46991
  -13.0676
  -2.87429
  -20.8263
......

NAN found. Exiting...
`

smcclusky, 2024-06-14:
Can you please send you actual .yaml configuration file and the Ginan software version you are using (the commit tag).

wsaddasw1989, 2024-06-14:
> Can you please send you actual .yaml configuration file and the Ginan software version you are using (the commit tag).

This is the git log  info of the software version.
commit 0f263270daa1d5234d1194d3bc314cf3dca85f38 (HEAD -> main, origin/main, origin/HEAD)
Merge: 7b79153 a92d809
Author: Ronald Maj <36747983+ronaldmaj@users.noreply.github.com>
Date:   Tue Apr 30 14:29:34 2024 +1000

I am using the default ppp_example.yaml file. And I don't think its the problem of yaml file.

`inputs:

    inputs_root: ./products/

    atx_files:  [ igs20.atx                             ] # required
    igrf_files: [ tables/igrf13coeffs.txt               ]
    erp_files:  [ finals.data.iau2000.txt               ]
    planetary_ephemeris_files: [ tables/DE436.1950.2050 ]

    troposphere:
        gpt2grid_files: [ tables/gpt_25.grd ]

    tides:
        ocean_tide_loading_blq_files:   [ tables/OLOAD_GO.BLQ                  ] # required if ocean loading is applied
        atmos_tide_loading_blq_files:   [ tables/ALOAD_GO.BLQ                  ] # required if atmospheric tide loading is applied
        ocean_pole_tide_loading_files:  [ tables/opoleloadcoefcmcor.txt ] # required if ocean pole tide loading is applied

    snx_files:
    # - "*.SNX" # use a wild card to include all files matching the description in the directory
    - igs_satellite_metadata.snx
    - tables/sat_yaw_bias_rate.snx
    - tables/qzss_yaw_modes.snx
    - tables/bds_yaw_modes.snx
    - IGS1R03SNX_20191950000_07D_07D_CRD.SNX

    satellite_data:
        # nav_files:
        # - "*.rnx"
        clk_files:
        # - "*.CLK"
        - IGS2R03FIN_20191990000_01D_30S_CLK.CLK
        bsx_files:
        # - "*.BIA"
        - IGS2R03FIN_20191990000_01D_01D_OSB.BIA
        sp3_files:
        # - "*.SP3"
        - IGS2R03FIN_20191990000_01D_05M_ORB.SP3

    gnss_observations:
        gnss_observations_root: ../data/
        rnx_inputs:
        # - "*.rnx"
        - ALIC00AUS_R_20191990000_01D_30S_MO.rnx
        - DARW00AUS_R_20191990000_01D_30S_MO.rnx
        - HOB200AUS_R_20191990000_01D_30S_MO.rnx
        - "M*.rnx"

outputs:

    metadata:
        config_description:    ppp_example

    outputs_root:              ./outputs/<CONFIG>

    gpx:
        output:                true
        filename:              <RECEIVER>_<CONFIG>_<YYYY><DDD><HH>.GPX

    trace:
        level:                 2
[... 204 lines cut ...]
            code_bias:
                estimated:          [true] # false
                sigma:              [20]
                process_noise:      [0]
            phase_bias:
                estimated:          [false]
                sigma:              [10]
                process_noise:      [0]

mongo:                   # Mongo is a database used to store results and intermediate values for later analysis and inter-process communication
    enable:              primary                     # Enable and connect to mongo database {none,primary,secondary,both}
    primary_uri:         mongodb://localhost:27017   # Location and port of the mongo database to connect to
    primary_database:    <CONFIG>
    primary_suffix:      ""                          # Suffix to append to database elements to make distinctions between runs for comparison
    # secondary_database:  <CONFIG>
    # secondary_suffix:    ""                          # Suffix to append to database elements to make distinctions between runs for comparison
    # secondary_uri:       mongodb://localhost:27017   # Location and port of the mongo database to connect to
    # output_config:       NONE                        # Output config {none,primary,secondary,both}
    output_components:   primary                     # Output components of measurements {none,primary,secondary,both}
    output_states:       primary                     # Output states {none,primary,secondary,both}
    output_measurements: primary                     # Output measurements and their residuals {none,primary,secondary,both}
    output_test_stats:   primary                     # Output test statistics {none,primary,secondary,both}
    delete_history:      primary                     # Drop the collection in the database at the beginning of the run to only show fresh data {none,primary,secondary,both}
    output_trace:        primary
`

smcclusky, 2024-06-16:
Hi wsaddasw1989,
We cannot reproduce your error on an AWS EC2 ubuntu 22.04 server using the same software version as you (same commit).
The  only thing that we can see is that the version of the software you are building is:
PEA starting... (main untagged-0f26327-dirty from Tue Apr 30 04:29:34 2024)

Whereas our fresh build from a github clone says:
PEA starting... (main untagged-0f26327 from Tue Apr 30 14:29:34 2024)

You can note the absence of "dirty" in our build, suggesting that something may have been changed in one of the Ginan directory monitored by git? Maybe you should try a rebuild to be sure something didn't get inadvertently changed?

wsaddasw1989, 2024-06-18:
Hi smcclusky,
The changed monitored by git  may caused by some unimportant modify.

I tried the ubuntu22.04.4  and the old ubuntu22.04, and change the linux kernal to the old version,  it just produce the same error.  This work was done on my local virtual machine.    But I when try it on the cloud ecs machine by unsing the ubuntu22.04 from the cloud service provider, it works well.  

Then I try ubuntu20.04.6 on my local virtual machine, it works well.  So I think there is something incompatible to the new Ubuntu22.04 version. But I don't know what cause it.

I am glad that I have bypassed this problem. Thank you very much.

closed 2024-06-21 by seballgeyer