GeoscienceAustralia/ginan #93: Question about source installation

JJaraG, 2024-06-24:
Dear Ginan team,

I am trying to install Ginan on Mac OSX, based on M1 max. I would like to do it using the source procedure stated on your webpage.

I have the following problem:

 jorge@build$ cmake ../
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- Checking whether C compiler has -isysroot
-- Checking whether C compiler has -isysroot - yes
-- Checking whether C compiler supports OSX deployment target flag
-- Checking whether C compiler supports OSX deployment target flag - yes
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /opt/homebrew/bin/gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Checking whether CXX compiler has -isysroot
-- Checking whether CXX compiler has -isysroot - yes
-- Checking whether CXX compiler supports OSX deployment target flag
-- Checking whether CXX compiler supports OSX deployment target flag - yes
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /opt/homebrew/bin/g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Setting ccache            off
-- Setting optimisation      on
-- Setting parallelisation   on
-- Found OpenMP_C: -fopenmp (found version "4.5")
-- Found OpenMP_CXX: -fopenmp (found version "4.5")
-- Found OpenMP: TRUE (found version "4.5")
-- Setting docs              on
-- Found Doxygen: /opt/homebrew/bin/doxygen (found version "1.11.0") found components: doxygen dot
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE
-- Found OpenSSL: /opt/homebrew/Cellar/openssl@3/3.3.1/lib/libcrypto.dylib (found version "3.3.1")
CMake Error at /opt/homebrew/lib/cmake/Boost-1.85.0/BoostConfig.cmake:141 (find_package):
  Could not find a package configuration file provided by
  "boost_stacktrace_backtrace" (requested version 1.85.0) with any of the
  following names:

    boost_stacktrace_backtraceConfig.cmake
    boost_stacktrace_backtrace-config.cmake

  Add the installation prefix of "boost_stacktrace_backtrace" to
  CMAKE_PREFIX_PATH or set "boost_stacktrace_backtrace_DIR" to a directory
  containing one of the above files.  If "boost_stacktrace_backtrace"
  provides a separate development package or SDK, be sure it has been
  installed.
Call Stack (most recent call first):
  /opt/homebrew/lib/cmake/Boost-1.85.0/BoostConfig.cmake:262 (boost_find_component)
  /opt/homebrew/Cellar/cmake/3.29.6/share/cmake/Modules/FindBoost.cmake:594 (find_package)
  CMakeLists.txt:147 (find_package)

-- Configuring incomplete, errors occurred!

Have you seen anything similar when installing on Mac (you said it has been tested).

Any help is very welcome!
Thanks in advance,

Best!
JJ

smcclusky, 2024-06-24:
Hi JJ,
Thanks for the question. Until recently all our Mac OSX testing had been done on OSX Intel (x86 silicon) systems, using only the native Clang Apple compiler. 

I have not seen your specific boost "boost_stacktrace_backtrace" problem with Ginan builds before, but I also have never tried to build using the GCC - GNU 13.2.0 compiler on OSX. I have seen mention of the Boost "boost_stacktrace_backtrace" libraries missing problem in stackoverflow. So it is not necessarily a Ginan specific problem I think.
 
Have you tried to find on your system the specific: `boost_stacktrace_backtraceConfig.cmake` and `boost_stacktrace_backtrace-config.cmake` and adding the appropriate prefix as suggested? You could possibly do that via CMake toolchain file as per below?

I have just been working on the Ginan Apple Silicon build problem myself last week as now I finally have a Apple Silicon Mac.
I have not done a lot of testing, but I have a solution that works on the latest OSX release (Sonoma 14.5) using the latest Ginan v3 `develop-weekly` snapshot and the native AppleClang compiler v15. Below is my recipe:

I use the Brew package manager to do the Ginan dependancy installation as per: 
`https://github.com/GeoscienceAustralia/ginan/blob/develop-weekly/scripts/installation/apple.md`

Ie: `brew install boost cmake eigen netcdf-cxx netcdf mongo-c-driver mongo-cxx-driver openblas openssl@3 yaml-cpp libomp`

Get the latest Ginan develop-weekly snapshot from GitHub: `https://github.com/GeoscienceAustralia/ginan/tree/develop-weekly`

Then from the ginan/src/build directory I do:
`cmake -DCMAKE_TOOLCHAIN_FILE=compile_mac_arm64.cmake ..`

This uses the new toolchain file (compile_mac_arm64.cmake ) that I needed to create to get CMake to find the brew installed OpenMP libraries and includes.

Let me know if any of this helps, else I will need to see if I can get the GCC compiler installed on my Mac to be able to reproduce your problem I think.

JJaraG, 2024-06-25:
Dear Simon,

Thanks a lot for the answer. Very kind and helpful to try to debug my installation problems. 

I tried to find the boost_stacktrace_backtraceConfig.cmake, but it did not work. It seems that in my Boost version (1.85), that file does not exist. I have tried with ***@***.*** installed using brew and I got similar issues. 

I have also tried the solution you provided with the develop_weekly. It passes without any problem the cmake, but when I make -j4, it fails with the errors in the attached log.file. 

If you can give me a hand passing the installation, it would be super nice. I am anxious to give a try, processing my GNSS networks using Ginan! 

Thanks in advance!
Best,
J
￼
—
Dr. Jorge Jara

MSCA Postdoctoral Fellow
Section 2.6 Seismic Hazard and Risk Dynamics
Helmholtz Centre Potsdam
GFZ German Research Centre For Geosciences
Telegrafenberg, D-14473 Potsdam

> On 25. Jun 2024, at 01:31, Simon McClusky ***@***.***> wrote:
> 
> cmake -DCMAKE_TOOLCHAIN_FILE=compile_mac_arm64.cmake ..

[  0%] Building CXX object cpp/CMakeFiles/otl.dir/loading/loading.cpp.o
[  0%] Building CXX object cpp/CMakeFiles/otl.dir/loading/tide.cpp.o
[  1%] Building CXX object cpp/CMakeFiles/otl.dir/loading/utils.cpp.o
[  1%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/a2af.c.o
[  1%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/a2tf.c.o
[  2%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/ab.c.o
[  2%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/ae2hd.c.o
[  2%] Building CXX object cpp/CMakeFiles/otl.dir/loading/load_functions.cpp.o
[  2%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/af2a.c.o
[  2%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/anp.c.o
[  3%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/anpm.c.o
[  3%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/apcg.c.o
[  3%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/apcg13.c.o
[  3%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/apci.c.o
[  4%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/apci13.c.o
[  4%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/apco.c.o
[  4%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/apco13.c.o
[  4%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/apcs.c.o
[  5%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/apcs13.c.o
[  5%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/aper.c.o
[  5%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/aper13.c.o
[  5%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/apio.c.o
In file included from /Users/jorgejara/PROGRAMS/GINAN/weekly/src/cpp/loading/tide.cpp:8:
In file included from /Users/jorgejara/PROGRAMS/GINAN/weekly/src/cpp/loading/tide.h:11:
In file included from /opt/homebrew/include/boost/multi_array.hpp:34:
In file included from /opt/homebrew/include/boost/multi_array/multi_array_ref.hpp:32:
/opt/homebrew/include/boost/functional.hpp:45:24: error: no member named 'unary_function' in namespace 'std'; did you mean '__unary_function'?
            using std::unary_function;
                  ~~~~~^~~~~~~~~~~~~~
                       __unary_function
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.5.sdk/usr/include/c++/v1/__functional/unary_function.h:46:1: note: '__unary_function' declared here
using __unary_function = __unary_function_keep_layout_base<_Arg, _Result>;
^
In file included from /Users/jorgejara/PROGRAMS/GINAN/weekly/src/cpp/loading/tide.cpp:8:
In file included from /Users/jorgejara/PROGRAMS/GINAN/weekly/src/cpp/loading/tide.h:11:
In file included from /opt/homebrew/include/boost/multi_array.hpp:34:
In file included from /opt/homebrew/include/boost/multi_array/multi_array_ref.hpp:32:
/opt/homebrew/include/boost/functional.hpp:46:24: error: no member named 'binary_function' in namespace 'std'; did you mean '__binary_function'?
            using std::binary_function;
                  ~~~~~^~~~~~~~~~~~~~~
                       __binary_function
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.5.sdk/usr/include/c++/v1/__functional/binary_function.h:49:1: note: '__binary_function' declared here
using __binary_function = __binary_function_keep_layout_base<_Arg1, _Arg2, _Result>;
[... 653 lines cut ...]
[ 58%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/tpsts.c.o
[ 58%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/tpxes.c.o
[ 58%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/tpxev.c.o
[ 58%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/tttai.c.o
[ 58%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/trxp.c.o
[ 58%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/trxpv.c.o
[ 59%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/tr.c.o
[ 59%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/ut1tai.c.o
[ 59%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/tttdb.c.o
[ 59%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/ttut1.c.o
[ 60%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/tttcg.c.o
[ 60%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/ut1utc.c.o
[... cut ...]
[ 62%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/xys00b.c.o
[ 62%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/xys06a.c.o
[ 63%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/zr.c.o
[ 63%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/zpv.c.o
[ 63%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/zp.c.o
[ 64%] Linking C static library /Users/jorgejara/PROGRAMS/GINAN/weekly/lib/libsofa_lib.a
[ 64%] Built target sofa_lib
make: *** [all] Error 2

JJaraG, 2024-06-25:
My apologies. I was able to install Ginan develop-weekly following the method you gave me, which are good news. I did a mistake in the Boost version I was using. Last one provided by Brew (1.85) works perfectly. I will proceed with the tests to check that everything is running properly. 

A small question before continuing: Every Friday you update Ginan in this Git branch, right? Should I update it every Friday, or is this option just in case I am processing data in real-time?

Thanks in advance,
Best,
Jorge

 
—
Dr. Jorge Jara

MSCA Postdoctoral Fellow
Section 2.6 Seismic Hazard and Risk Dynamics
Helmholtz Centre Potsdam
GFZ German Research Centre For Geosciences
Telegrafenberg, D-14473 Potsdam

> On 25. Jun 2024, at 13:44, Jorge Jara ***@***.***> wrote:
> 
> Dear Simon,
> 
> Thanks a lot for the answer. Very kind and helpful to try to debug my installation problems. 
> 
> I tried to find the boost_stacktrace_backtraceConfig.cmake, but it did not work. It seems that in my Boost version (1.85), that file does not exist. I have tried with ***@***.*** installed using brew and I got similar issues. 
> 
> I have also tried the solution you provided with the develop_weekly. It passes without any problem the cmake, but when I make -j4, it fails with the errors in the attached log.file. 
> 
> If you can give me a hand passing the installation, it would be super nice. I am anxious to give a try, processing my GNSS networks using Ginan! 
> 
> Thanks in advance!
> Best,
> J
> <log.txt>
> 
> —
> Dr. Jorge Jara
> 
> MSCA Postdoctoral Fellow
> Section 2.6 Seismic Hazard and Risk Dynamics
> Helmholtz Centre Potsdam
> GFZ German Research Centre For Geosciences
> Telegrafenberg, D-14473 Potsdam
> 
>> On 25. Jun 2024, at 01:31, Simon McClusky ***@***.***> wrote:
>> 
>> cmake -DCMAKE_TOOLCHAIN_FILE=compile_mac_arm64.cmake ..
>

polytopes-design, 2024-06-26:
The Friday updates will usually contain any improvements that have been made to the source code over the previous week. It is not necessary to update for real-time, but feel free to pull the updates and new features as they become available.

JJaraG, 2024-06-26:
Thanks a lot! The help has been very useful

Best,
Jorge

closed 2024-07-02 by seballgeyer