GeoscienceAustralia/ginan #99: Error in Niell tropo function calculation ?

jonathanmuller, 2024-07-29:
Hi, 

There is a common pattern I see in all implementations of the Niell tropo across github :

The implementation is always the same : 

`(1+a / ... ) / (sinel +b / ...)` while I believe it should be ` (sinel +b / ...) / (1+a / ... ) ` (the opposite)

For example in the code : https://github.com/GeoscienceAustralia/ginan/blob/0f263270daa1d5234d1194d3bc314cf3dca85f38/src/cpp/trop/tropModels.cpp#L119

```
double mapHerring(double el, double a, double b, double c)
{
	double sinel = sin(el);
	return	(1 + a /
				(1 + b /
					(1 + c))) / (sinel + (a /
										(sinel + b /
												(sinel + c))));
}
```

However, original paper ( https://safe.nrao.edu/wiki/pub/Main/RefBendDelayCalc/Niell_Global.pdf ) list "equation 4" as :

[image]
So the form is `(1/(1+a / ... )) / (1+1/(sinel +b / ...))`
As `(1/y)/(1/x)` can be rewritten as `x/y`
Then the calculation above can be simplified as `(sinel +b / ...) /(1+a / ... ) `
There can be no ambiguity as later it is used directly as follow, without further division or calculation directly as height correction : 
https://github.com/GeoscienceAustralia/ginan/blob/0f263270daa1d5234d1194d3bc314cf3dca85f38/src/cpp/trop/tropGPT2.cpp#L83
[image]

This is the same (wrong?) implementation as RTKLib and any other commonly used RTK library I could find on github.
Am I just wrong about the understanding of the calculation, or is everyone copy-pasting the same error ?

seballgeyer, 2024-08-16:
Hello, 

Thank you for your observation regarding the implementation of the Niell tropospheric model in our code. We appreciate your thorough examination and the references provided, particularly your comparison with the original Niell (1996) paper and the subsequent use of the model in our software.

It's important to note that the original Niell (1996) paper contains an error in the equation. The correct form of the equation is indeed as it is implemented in the code, and this is supported by earlier works such as Herring (1992) and subsequent papers that have clarified the correct implementation (like Niell 2000, https://doi.org/10.1186/BF03352267). This discrepancy might have led to some confusion in the implementations you’ve seen across GitHub, but rest assured that the approach we have taken in our code aligns with the corrected and widely accepted form.

We recommend continuing with the implementation as it currently stands in our software, as it reflects the correct interpretation of the tropospheric mapping function. If you have any further questions or need additional clarification, please feel free to reach out. Your attention to detail is greatly appreciated, and we're always here to support the community in ensuring the highest accuracy and integrity of the code.

Best regards, 
Sébastien

closed 2024-08-16 by seballgeyer

jonathanmuller, 2024-08-21:
You fully convinced me that you -and the code- where right
Thank you a lot for all the details !