GeoscienceAustralia/ginan #102: we need your table files

LJC8848, 2024-09-07:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Configuration files**
Yaml files assist with with confirming the exact processing options of the software.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Operating environment (please complete the following information):**
- OS: [e.g. Widows, Linux, Mac + Version]
- Version of Ginan:
- Library used (Boost, Eigen, ...)

These may be found in the 'Compilation details' section of the pea's output.

**Additional context**
Add any other context about the problem here.

seballgeyer, 2024-09-07:
Thanks for reporting this issue!

We are aware that users in China may experience connectivity issues when accessing files hosted on AWS S3, likely due to regional network restrictions that affect AWS services in certain locations.

We're exploring solutions to ensure all users have reliable access to the necessary files.

We’ll keep you updated as we work towards a permanent fix.

Sébastien

LJC8848, 2024-09-07:
#- tables/sat_yaw_bias_rate.snx
        #- tables/qzss_yaw_modes.snx
        #- tables/bds_yaw_modes.snx
could you send these files to me ？  I need them .  email: 206258902@qq.com

LJC8848, 2024-09-07:
> Thanks for reporting this issue!感谢您报告此问题！
> 
> We are aware that users in China may experience connectivity issues when accessing files hosted on AWS S3, likely due to regional network restrictions that affect AWS services in certain locations.我们知道，中国用户在访问 AWS S3 上托管的文件时可能会遇到连接问题，这可能是由于区域网络限制影响了某些位置的 AWS 服务。
> 
> We're exploring solutions to ensure all users have reliable access to the necessary files.我们正在探索解决方案，以确保所有用户都能可靠地访问必要的文件。
> 
> We’ll keep you updated as we work towards a permanent fix.在我们努力寻求永久修复的过程中，我们会及时通知您。
> 
> Sébastien 塞巴斯蒂安
#- tables/sat_yaw_bias_rate.snx
#- tables/qzss_yaw_modes.snx
#- tables/bds_yaw_modes.snx
could you send these files to me ？ I need them . email: [206258902@qq.com](mailto:206258902@qq.com)

LJC8848, 2024-09-07:
> Thanks for reporting this issue!
>
> We are aware that users in China may experience connectivity issues when accessing files hosted on AWS S3, likely due to regional network restrictions that affect [...]

Sébastien 
#- tables/sat_yaw_bias_rate.snx
#- tables/qzss_yaw_modes.snx
#- tables/bds_yaw_modes.snx
could you send these files to me ？ I need them . email: [206258902@qq.com](mailto:206258902@qq.com)

LJC8848, 2024-09-07:
> Thanks for reporting this issue!
>
> We are aware that users in China may experience connectivity issues when accessing files hosted on AWS S3, likely due to regional network restrictions that affect [...]

If possible, I need all the dependent files contained in the table in the example program. I should be able to find other products and data elsewhere.
Excuse me, will you release the PPP - RTK code next month? There is news saying that the source code of PPP - RTK will be released around October. Is it true?