GeoscienceAustralia/ginan #103: Problems encountered during operation

LJC8848, 2024-09-07:
PEA starting... (  from )

Checking configuration file ppp_example.yaml
Loading configuration from file ppp_example.yaml

===============================
Configuration...
===============================
Inputs:
	snx_files:                       ./products/tables/igs_satellite_metadata.snx ./products/igs19P2062.SNX 
	atx_files:                       ./products/tables/igs20.atx 
	clk_files:                       ./products/igs20624.CLK 
	bsx_files:                       ./products/COD0MGXFIN_20191990000_01D_01D_OSB.BIA 
	igrf_files:                      ./products/tables/igrf13coeffs.txt 
	ocean_tide_loading_blq_files:    ./products/tables/OLOAD_GO.BLQ 
	atmos_tide_loading_blq_files:    ./products/tables/ALOAD_GO.BLQ 
	ocean_pole_tide_loading_files:   ./products/tables/opoleloadcoefcmcor.txt 
	erp_files:                       ./products/tables/finals.data.iau2000.txt 
	sp3_files:                       ./products/igs20624.SP3 
	planetary_ephemeris_files:       ./products/tables/DE436.1950.2050 
	gpt2grid_files:                  ./products/tables/gpt_25.grd 
	rnx_inputs:                      ./products/../data/ALIC00AUS_R_20191990000_01D_30S_MO.rnx ./products/../data/DARW00AUS_R_20191990000_01D_30S_MO.rnx ./products/../data/HOB200AUS_R_20191990000_01D_30S_MO.rnx 

Outputs:
	trace level:                   2
	receiver trace filename:       ./outputs/ppp_example/<RECEIVER>_ppp_example_<YYYY><DDD><HH>.TRACE
	network trace filename:        ./outputs/ppp_example/<RECEIVER>_ppp_example_<YYYY><DDD><HH>.TRACE
	gpx filename:                  ./outputs/ppp_example/<RECEIVER>_ppp_example_<YYYY><DDD><HH>.GPX

Process Modes:
	Preprocessor:        1
	SPP                  1
	PPP:                 1
	Minimum Constraints: 0
	Ionospheric:         1
	RTS Smoothing:       1

Systems:
	GPS:     1
	GLONASS: 0
	GALILEO: 1
	BEIDOU:  0
	QZSS:    0
	LEO:     0

Epochs:
	epoch_interval: 30

===============================
...End Configuration
===============================

Threading with max 12 eigen threads
Threading with max 2147483647 omp threads
Logging with trace level:2

Mongo connecting to database @ mongodb://localhost:27017
Loading ATX file ./products/tables/igs20.atx
Loading SP3 file ./products/igs20624.SP3
Loading ERP file ./products/tables/finals.data.iau2000.txt
Loading CLK file ./products/igs20624.CLK
Loading BSX file ./products/COD0MGXFIN_20191990000_01D_01D_OSB.BIA
Loading IGRF file ./products/tables/igrf13coeffs.txt
Loading SNX file ./products/tables/igs_satellite_metadata.snx
Loading SNX file ./products/igs19P2062.SNX
Loading GPT file ./products/tables/gpt_25.grd
Loading Ocean Pole Tide file ./products/tables/opoleloadcoefcmcor.txt
Loading planetary ephemeris file ./products/tables/DE436.1950.2050
Initialising station ALIC
Initialising station DARW
[... 116 lines cut ...]
Warning: Block type not found for E32, attitude modelling etc may be affected, check sinex file
Warning: Looking for precise position, but no precise ephemerides found for E32
Warning: No sat pos found for E32.
Warning: Looking for precise position, but no precise ephemerides found for E33
Warning: No sat pos found for E33.
Warning: SVN not found for E34
Warning: Block type not found for E34, attitude modelling etc may be affected, check sinex file
Warning: Looking for precise position, but no precise ephemerides found for E34
Warning: No sat pos found for E34.
Warning: SVN not found for E35
Warning: Block type not found for E35, attitude modelling etc may be affected, check sinex file
Warning: Looking for precise position, but no precise ephemerides found for E35
Warning: No sat pos found for E35.
Warning: Looking for precise position, but no precise ephemerides found for E36
Warning: No sat pos found for E36.
 ------- PREPROCESSING STATIONS       --------

 ------- DOING STATE TRANSITION       --------

 ------- CALCULATING PPP MEASUREMENTS --------

Warning: eop is estimated but model is not enabledterminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 18446744071562067968) >= this->size() (which is 731)
已中止 (核心已转储)
[ppp_example.txt](https://github.com/user-attachments/files/16918698/ppp_example.txt)

seballgeyer, 2024-09-12:
Same error as issue 104

closed 2024-09-12 by seballgeyer

LJC8848, 2024-09-12:
../bin/pea --config rt_ppp_example.yaml

PEA starting... (&nbsp; from )

Checking configuration file rt_ppp_example.yaml
Loading configuration from file rt_ppp_example.yaml

===============================
Configuration Summary...
===============================
Inputs:
&nbsp;&nbsp;&nbsp; snx_files:&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; ./products/tables/igs_satellite_metadata_2203_plus.snx 
&nbsp;&nbsp;&nbsp; atx_files:&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; ./products/tables/igs20.atx 
&nbsp;&nbsp;&nbsp; igrf_files:&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; ./products/tables/igrf13coeffs.txt 
&nbsp;&nbsp;&nbsp; ocean_tide_loading_blq_files:&nbsp;&nbsp;&nbsp; ./products/tables/OLOAD_GO.BLQ 
&nbsp;&nbsp;&nbsp; atmos_tide_loading_blq_files:&nbsp;&nbsp;&nbsp; ./products/tables/ALOAD_GO.BLQ 
&nbsp;&nbsp;&nbsp; ocean_pole_tide_loading_files:&nbsp;&nbsp; ./products/tables/opoleloadcoefcmcor.txt 
&nbsp;&nbsp;&nbsp; erp_files:&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; ./products/tables/finals.data.iau2000.txt 
&nbsp;&nbsp;&nbsp; egm_files:&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; ./products/tables/EGM2008.gfc 
&nbsp;&nbsp;&nbsp; planetary_ephemeris_files:&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; ./products/tables/DE436.1950.2050 
&nbsp;&nbsp;&nbsp; ocean_tide_potential_files:&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; ./products/tables/fes2014b_Cnm-Snm.dat 
&nbsp;&nbsp;&nbsp; gpt2grid_files:&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; ./products/tables/gpt_25.grd 
&nbsp;&nbsp;&nbsp; rtcm_inputs:&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; ***@***.******@***.***/BCEP00BKG0 ***@***.******@***.***/SSRA00BKG0 
&nbsp;&nbsp;&nbsp; rtcm_inputs:&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; ***@***.******@***.***/ALIC00AUS0 ***@***.******@***.***/STR200AUS0 

Outputs:
&nbsp;&nbsp;&nbsp; trace level:&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; 4
&nbsp;&nbsp;&nbsp; receiver trace filename:&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; ./outputs/rt_ppp_example/<RECEIVER&gt;_rt_ppp_example_<YYYY&gt;<DDD&gt;<HH&gt;.TRACE
&nbsp;&nbsp;&nbsp; network trace filename:&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; ./outputs/rt_ppp_example/<RECEIVER&gt;_rt_ppp_example_<YYYY&gt;<DDD&gt;<HH&gt;.TRACE
&nbsp;&nbsp;&nbsp; gpx filename:&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; ./outputs/rt_ppp_example/rt_ppp_example_<RECEIVER&gt;_<YYYY&gt;<DDD&gt;<HH&gt;.GPX

Process Modes:
&nbsp;&nbsp;&nbsp; Preprocessor:&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; 1
&nbsp;&nbsp;&nbsp; SPP&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; 1
&nbsp;&nbsp;&nbsp; PPP:&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; 1
&nbsp;&nbsp;&nbsp; Minimum Constraints: 0
&nbsp;&nbsp;&nbsp; Ionospheric:&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; 0
&nbsp;&nbsp;&nbsp; RTS Smoothing:&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; 0

Systems:
&nbsp;&nbsp;&nbsp; GPS:&nbsp;&nbsp;&nbsp;&nbsp; 1
&nbsp;&nbsp;&nbsp; GLONASS: 0
&nbsp;&nbsp;&nbsp; GALILEO: 0
&nbsp;&nbsp;&nbsp; BEIDOU:&nbsp; 0
&nbsp;&nbsp;&nbsp; QZSS:&nbsp;&nbsp;&nbsp; 0
&nbsp;&nbsp;&nbsp; LEO:&nbsp;&nbsp;&nbsp;&nbsp; 0

Epochs:
&nbsp;&nbsp;&nbsp; epoch_interval: 20

===============================
...End Configuration Summary
===============================

Compilation details:
====================
Ginan branch:&nbsp;&nbsp;&nbsp; &nbsp;
Ginan version:&nbsp;&nbsp; &nbsp;
Commit date:&nbsp;&nbsp;&nbsp;&nbsp; &nbsp;
Operating system: "Ubuntu 22.04.4 LTS"
Compiler version: GNU 11.4.0
Boost version:&nbsp;&nbsp;&nbsp; 1.74.0
Eigen version:&nbsp;&nbsp;&nbsp; 3.4.0
Mongocxx version: 3.7.0

Runtime details:
================
Logging at trace level4
Threading with max 16 eigen threads
Threading with max 16 omp threads

Mongo connecting to database @ mongodb://localhost:27017
Loading ATX file ./products/tables/igs20.atx
Loading ERP file ./products/tables/finals.data.iau2000.txt
Loading IGRF file ./products/tables/igrf13coeffs.txt
Loading SNX file ./products/tables/igs_satellite_metadata_2203_plus.snx
Loading GPT file ./products/tables/gpt_25.grd
Loading Ocean Pole Tide file ./products/tables/opoleloadcoefcmcor.txt
Loading EGM file ./products/tables/EGM2008.gfc
Loading Tide file ./products/tables/fes2014b_Cnm-Snm.dat
Loading planetary ephemeris file ./products/tables/DE436.1950.2050
Initialising station ALIC
Initialising station STR2
Mongo connecting to database : rt_ppp_example @ mongodb://localhost:27017
Mongo connected&nbsp; to database : rt_ppp_example

Starting to process epochs...

Starting epoch #1
Synced 0 receivers...
Processed epoch #1 - GPS time: 2331 362740 - 2024-Sep-12 04:45:40 (took 00:00:00.00)

Starting epoch #2
Synced 2 receivers...
&nbsp;------- PREPROCESSING STATIONS&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; --------
double free or corruption (out)
已中止 (核心已转储)