GeoscienceAustralia/ginan #104: Container overflow problem

LJC8848, 2024-09-11:
PEA starting... (  from )

Checking configuration file ppp_example.yaml
Loading configuration from file ppp_example.yaml

===============================
Configuration Summary...
===============================
Inputs:
	snx_files:                       ./products/tables/igs_satellite_metadata.snx ./products/tables/sat_yaw_bias_rate.snx ./products/tables/qzss_yaw_modes.snx ./products/tables/bds_yaw_modes.snx ./products/IGS0OPSSNX_20240010000_01D_01D_CRD.SNX 
	atx_files:                       ./products/tables/igs20.atx 
	clk_files:                       ./products/IGS0OPSFIN_20240010000_01D_30S_CLK.CLK 
	bsx_files:                       ./products/CAS0MGXRAP_20240010000_01D_01D_OSB.BIA 
	igrf_files:                      ./products/tables/igrf13coeffs.txt 
	ocean_tide_loading_blq_files:    ./products/tables/OLOAD_GO.BLQ 
	atmos_tide_loading_blq_files:    ./products/tables/ALOAD_GO.BLQ 
	ocean_pole_tide_loading_files:   ./products/tables/opoleloadcoefcmcor.txt 
	erp_files:                       ./products/tables/finals.data.iau2000.txt 
	sp3_files:                       ./products/IGS0OPSFIN_20240010000_01D_15M_ORB.SP3 
	planetary_ephemeris_files:       ./products/tables/DE436.1950.2050 
	gpt2grid_files:                  ./products/tables/gpt_25.grd 
	rnx_inputs:                      ./products/../data/ABMF00GLP_R_20240010000_01D_30S_MO.rnx 

Outputs:
	trace level:                   2
	receiver trace filename:       ./outputs/ppp_example/<RECEIVER>_ppp_example_<YYYY><DDD><HH>.TRACE
	network trace filename:        ./outputs/ppp_example/<RECEIVER>_ppp_example_<YYYY><DDD><HH>.TRACE
	gpx filename:                  ./outputs/ppp_example/<RECEIVER>_ppp_example_<YYYY><DDD><HH>.GPX
	pos filename:                  ./outputs/ppp_example/<RECEIVER>_ppp_example_<YYYY><DDD><HH>.POS

Process Modes:
	Preprocessor:        1
	SPP                  1
	PPP:                 1
	Minimum Constraints: 0
	Ionospheric:         0
	RTS Smoothing:       1

Systems:
	GPS:     1
	GLONASS: 0
	GALILEO: 0
	BEIDOU:  0
	QZSS:    0
	LEO:     0

Epochs:
	epoch_interval: 30
	max_epochs:     2880
	epoch start:    2024-Jan-01 00:00:00
	epoch end:      2024-Jan-01 23:59:30

===============================
...End Configuration Summary
===============================

Compilation details:
====================
Ginan branch:     
Ginan version:    
Commit date:      
Operating system: "Ubuntu 22.04.4 LTS"
Compiler version: GNU 11.4.0
Boost version:    1.74.0
Eigen version:    3.4.0
Mongocxx version: 3.7.0

Runtime details:
================
Logging at trace level2
Threading with max 16 eigen threads
Threading with max 16 omp threads

Mongo connecting to database @ mongodb://localhost:27017
Loading ATX file ./products/tables/igs20.atx
Loading SP3 file ./products/IGS0OPSFIN_20240010000_01D_15M_ORB.SP3
Loading ERP file ./products/tables/finals.data.iau2000.txt
Loading CLK file ./products/IGS0OPSFIN_20240010000_01D_30S_CLK.CLK
Loading BSX file ./products/CAS0MGXRAP_20240010000_01D_01D_OSB.BIA
Loading IGRF file ./products/tables/igrf13coeffs.txt
Loading SNX file ./products/tables/igs_satellite_metadata.snx
Loading SNX file ./products/tables/sat_yaw_bias_rate.snx
Loading SNX file ./products/tables/qzss_yaw_modes.snx
Loading SNX file ./products/tables/bds_yaw_modes.snx
Loading SNX file ./products/IGS0OPSSNX_20240010000_01D_01D_CRD.SNX
Loading GPT file ./products/tables/gpt_25.grd
Loading Ocean Pole Tide file ./products/tables/opoleloadcoefcmcor.txt
Loading planetary ephemeris file ./products/tables/DE436.1950.2050
Initialising station ABMF
Mongo connecting to database : ppp_example @ mongodb://localhost:27017
Mongo connected  to database : ppp_example

Starting to process epochs...

Starting epoch #1
Synced 1 receivers...
 ------- PREPROCESSING STATIONS       --------

Warning: No sat clk found for G27.
Warning: No suitable data for bias interpolation, extrapolated bias in use.
Warning: No receiver apriori clock found for ABMF
 ------- DOING STATE TRANSITION       --------

 ------- CALCULATING PPP MEASUREMENTS --------
terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 18446744071562067968) >= this->size() (which is 731)
已中止 (核心已转储)
Warning: Excessive time elapsed, skipping epoch 1. Configuration 'wait_next_epoch' is 0

seballgeyer, 2024-09-11:
Hello, 

It looks like the error you're encountering is:

```
terminate called after throwing an instance of 'std::out_of_range'
what(): vector::_M_range_check: __n (which is 18446744071562067968) >= this->size() (which is 731)
```

This means the code is attempting to access an index in a std::vector that is out of range. Specifically, the index (18446744071562067968) is much larger than the vector's size (731).

This kind of issue typically occurs when the input data or a parameter passed to the code is incorrect or unexpected. It can happen if the data structure you're working with is not initialized properly or if there's a bug in how you're reading or processing the input file. Given the size mismatch, the issue may be related to how the input file is being handled.

Could you provide the full config file you're using, and all other input files or point to where you got it from? Right now, looking at the content of the log, there is mostly the `CAS0MGXRAP_20240010000_01D_01D_OSB.BIA` that we don't know where it comes from. That way, we can trace the issue back to its source and check if there's something wrong with the data or how it's being processed.

Best regards,
Sebastien

LJC8848, 2024-09-11:
Hello:Sebastien
During the usage process, if I use GDB for debugging and running, a PEA  runtime error occurred in version main, as shown in ginan - main - GDB in  the bug folder. At this time, my local time is around 21:00 Beijing  time, while the running display shows that PEA started processing at  2024 - 09 - 11 13:47:33.32 and finished processing at 2024 - 09 - 11  13:47:33.33. I think there might be problems with time alignment, data  reading, and other aspects. When I used the ginan - 3.1 version for  debugging with GDB, an epochs loop as shown in the pictures in the bug  folder occurred. I don't know if it is a problem with my configuration  or a bug in the software. I used the data from the ginan_aux - main  routine and also encountered file reading problems. I have sent you my  installation package and the bug together, hoping that the problem can  be found. In addition, I would like to know if the source code of PPP -  RTK will be made public next month or in the near future. Will it be  based on ginan PPP or another software? Looking forward to your reply.

李金成
***@***.***

 

 
 terminate called after throwing an instance of 'std::out_of_range' what(): vector::_M_range_check: __n (which is 18446744071562067968) &gt;= this-&gt;size() (which is 731)  
 
 
 
 

从QQ邮箱发来的超大附件

inputData.zip (111.36M, 2024年10月11日 20:21 到期)进入下载页面：https://mail.qq.com/cgi-bin/ftnExs_download?k=7338303586063ccce37a7b394433041a4d4c09035a070105180c5557541e07060c0a1d0657500018000c56025b055202055d08006222365c5b484541265242541510021c4c495f453505&t=exs_ftn_download&code=5805b365

ginan-main.zip (1.10G, 2024年10月11日 22:52 到期)进入下载页面：https://mail.qq.com/cgi-bin/ftnExs_download?k=7a3762664f91fdddef75296a4039044b41435052510d5400140750535f1454540d074f54530b07490e050057530e57050951010266373603505903084b54570d5719180f16390b&t=exs_ftn_download&code=97bff96d

ginan3.1.zip (747.21M, 2024年10月11日 22:52 到期)进入下载页面：https://mail.qq.com/cgi-bin/ftnExs_download?k=27343736a1992fb7b0767c3a403451491e40030405065b514b57520f5619520751001a055f57004b510353530503535251070652663863010f5a5658551a52481c5d47365b&t=exs_ftn_download&code=f476f4cf

bug.zip (2.22M, 无限期)进入下载页面：https://mail.qq.com/cgi-bin/ftnExs_download?k=046637669150b8cfee247c6a4436561f401252525353560515030200011b0607085e1a000454551d0a570f030155510659550251623164524d01191c0b46640d&t=exs_ftn_download&code=8f7fb6d0

seballgeyer, 2024-09-12:
Hello

It looks like the issue is caused by using the wrong input file. Specifically, you've used opoleloadcmcor.txt, which is the example time series file, instead of the correct grid file. The error you're seeing is likely a result of this mismatch between the input format and what the code expects.

The time series file contains data in a format that's incompatible with the grid file format required by the program. This leads to the code trying to process the data incorrectly, causing the out-of-range error you encountered. Since the file structures are different, using the wrong file results in vector access errors.

To resolve the issue, you'll need to replace opoleloadcmcor.txt with the correct file, which contains the necessary grid data. You can download the correct file from this link [https://iers-conventions.obspm.fr/content/chapter7/additional_info/opoleloadcoefcmcor.txt.gz], and uncompress it. Once you've swapped out the files, the code should be able to process the input correctly. We will work in the near future on better check that the input files are those that Ginan expects. 

Regarding your other concern:
- Time in the log file: Ginan writes all times in UTC not local time (9pm in Beijing is 2pm UTC)
- We are unaware of the source code of PPP-RTK you mentioned, so I believed it won't be based on Ginan

Best regards,
Sébastien

LJC8848, 2024-09-12:
Hello:
By downloading this file, I have solved these problems that occurred. Thank you very much.

Best wishes,
Jincheng

?????
***@***.***

closed 2024-09-12 by seballgeyer