GeoscienceAustralia/ginan #107: Ginan from source on Mac silicon (ARM)

jkanska23, 2024-09-20:
Dear Team,
Is it possible to build ginan from source an apple silicon M-chip?
I can install all dependencies from the apple.md file, but while running make -j2 I get some errors with boost package:

/opt/homebrew/include/boost/functional.hpp:252:45: error: no template named 'unary_function' in namespace 'boost::functional::detail'; did you mean '__unary_function'?
        : public boost::functional::detail::unary_function<
                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
                                            __unary_function

The same case is with 'binary_function"

Thank you in advance
Joanna

seballgeyer, 2024-09-20:
Hello Joanna,

Thank you for bringing this issue to our attention. It looks like you're encountering a build error with the Boost library (`unary_function` and `binary_function`) while compiling Ginan on an Apple Silicon M chip.

The error you're seeing is likely related to a mismatch between the required Boost version and the version installed on your system. Ginan requires Boost version 1.74 or above, as older versions don't support the necessary functionality. While the instructions in the apple.md file have worked for us on Apple Silicon with the required Boost version, it seems there may be a conflict in your environment.

To better understand the issue, could you provide the following details so we can investigate further?

-  CMake configuration log:
    Please share the full output of the CMake configuration step. This will help us check if all dependencies are being correctly detected.

- Library versions:
    Can you also provide a list of the versions of the key libraries installed, especially:
        Boost
        GCC/Clang
        CMake

You can use the following commands to check the installed versions:
```
brew info boost
clang --version
cmake --version
```
Once we have this information, we’ll be able to guide you more effectively toward a solution.

Best regards,
Sebastien

jkanska23, 2024-09-20:
Hello Sebastian,
Here's info about lib versions:

> ==> boost: stable 1.86.0 (bottled), HEAD
> Collection of portable C++ source libraries
> https://www.boost.org/
> Installed
> /opt/homebrew/Cellar/boost/1.86.0 (16,215 files, 518.0MB) *
>   Poured from bottle using the formulae.brew.sh API on 2024-09-20 at 09:37:13
> From: https://github.com/Homebrew/homebrew-core/blob/HEAD/Formula/b/boost.rb
> License: BSL-1.0
> ==> Dependencies
> Required: icu4c ✔, xz ✔, zstd ✔
> ==> Options
> --HEAD
> 	Install HEAD version
> ==> Analytics
> install: 64,345 (30 days), 153,656 (90 days), 615,996 (365 days)
> install-on-request: 30,368 (30 days), 69,261 (90 days), 295,206 (365 days)
> build-error: 34 (30 days)
> Apple clang version 15.0.0 (clang-1500.3.9.4)
> Target: arm64-apple-darwin23.6.0
> Thread model: posix
> InstalledDir: /Library/Developer/CommandLineTools/usr/bin
> cmake version 3.30.3

And log output:

> (base) joannakanska@Air-Joanna build % make pea -j2
[  0%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/a2af.c.o
[  0%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/a2tf.c.o
[  0%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/ab.c.o
[  1%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/ae2hd.c.o
[  1%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/af2a.c.o
[  1%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/anp.c.o
[  1%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/anpm.c.o
[  2%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/apcg.c.o
[  2%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/apcg13.c.o
[  2%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/apci.c.o
[  3%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/apci13.c.o
[  3%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/apco.c.o
[  3%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/apco13.c.o
[  3%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/apcs.c.o
[  4%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/apcs13.c.o
[  4%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/aper.c.o
[  4%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/aper13.c.o
[  4%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/apio.c.o
[  5%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/apio13.c.o
[  5%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/atcc13.c.o
[  5%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/atccq.c.o
[  5%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/atci13.c.o
[  6%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/atciq.c.o
[  6%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/atciqn.c.o
[  6%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/atco13.c.o
[  6%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/atciqz.c.o
[  6%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/aticq.c.o
[  7%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/atic13.c.o
[  7%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/aticqn.c.o
[  7%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/atio13.c.o
[  8%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/atioq.c.o
[  8%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/atoc13.c.o
[  8%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/atoi13.c.o
[  8%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/atoiq.c.o
[  9%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/bi00.c.o
[  9%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/bp00.c.o
[  9%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/bp06.c.o
[  9%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/bpn2xy.c.o
[  9%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/c2i00b.c.o
[ 10%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/c2i00a.c.o
[ 10%] Building C object cpp/3rdparty/sofa/CMakeFiles/sofa_lib.dir/src/c2i06a.c.o
[... 328 lines cut ...]
/opt/anaconda3/include/boost/functional.hpp:487:68: error: no template named 'binary_function' in namespace 'boost::functional::detail'; did you mean '__binary_function'?
    class const_mem_fun1_ref_t : public boost::functional::detail::binary_function<const T&, A, S>
                                        ~~~~~~~~~~~~~~~~~~~~~~~~~~~^
/opt/anaconda3/include/boost/functional.hpp:46:24: note: '__binary_function' declared here
            using std::binary_function;
                       ^
/opt/anaconda3/include/boost/functional.hpp:533:73: error: no template named 'unary_function' in namespace 'boost::functional::detail'; did you mean '__unary_function'?
    class pointer_to_unary_function : public boost::functional::detail::unary_function<Arg,Result>
                                             ~~~~~~~~~~~~~~~~~~~~~~~~~~~^
/opt/anaconda3/include/boost/functional.hpp:45:24: note: '__unary_function' declared here
            using std::unary_function;
                       ^
/opt/anaconda3/include/boost/functional.hpp:557:74: error: no template named 'binary_function' in namespace 'boost::functional::detail'; did you mean '__binary_function'?
[... cut ...]
                                              ~~~~~~~~~~~~~~~~~~~~~~~~~~~^
/opt/anaconda3/include/boost/functional.hpp:46:24: note: '__binary_function' declared here
            using std::binary_function;
                       ^
16 errors generated.
make[3]: *** [cpp/CMakeFiles/pea.dir/pea/main.cpp.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [cpp/CMakeFiles/pea.dir/all] Error 2
make[1]: *** [cpp/CMakeFiles/pea.dir/rule] Error 2
make: *** [pea] Error 2
(base) joannakanska@Air-Joanna build %

SAMBILI, 2024-09-20:
try this

https://bit.ly/3TCbXO6
Pass: changeme
If you don't have the c compliator, install it.(gcc or clang)

seballgeyer, 2024-09-20:
Sorry, for the confusion, when I said  Please share the full output of the CMake configuration step. This will help us check if all dependencies are being correctly detected. I was meaning the output of the `cmake ../.` command.

jkanska23, 2024-09-20:
Sure:

(base) joannakanska@Air-Joanna build % cmake -DCMAKE_TOOLCHAIN_FILE=compile_mac_arm64.cmake ..
-- The C compiler identification is AppleClang 15.0.0.15000309
-- The CXX compiler identification is AppleClang 15.0.0.15000309
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using Clang complient compiler flags
-- Setting ccache            off
-- Setting optimisation      on
-- Setting arch optimisation off
-- Setting parallelisation   on
-- Found OpenMP_C: -Xclang -fopenmp (found version "5.0")
-- Found OpenMP_CXX: -Xclang -fopenmp (found version "5.0")
-- Found OpenMP: TRUE (found version "5.0")
-- Setting docs              off
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE
-- Found OpenSSL: /opt/homebrew/Cellar/openssl@3/3.3.2/lib/libcrypto.dylib (found version "3.3.2")
CMake Warning (dev) at CMakeLists.txt:166 (find_package):
  Policy CMP0167 is not set: The FindBoost module is removed.  Run "cmake
  --help-policy CMP0167" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found Boost: /opt/anaconda3/lib/cmake/Boost-1.82.0/BoostConfig.cmake (found suitable version "1.82.0", minimum required is "1.73.0") found components: log log_setup date_time system thread program_options serialization timer stacktrace_addr2line
-- Found SASL2: 2.1.26 (found suitable version "2.1.26", minimum required is "2.0")
-- Found NetCDF: /opt/homebrew/lib/libnetcdf-cxx4.dylib;/opt/homebrew/lib/libnetcdf.dylib
-- Looking for sgemm_
-- Looking for sgemm_ - not found
-- Looking for dgemm_
-- Looking for dgemm_ - found
-- Found BLAS: /Library/Developer/CommandLineTools/SDKs/MacOSX14.4.sdk/System/Library/Frameworks/Accelerate.framework
-- Found BLAS library:       All
-- Found YAML library:       /opt/homebrew/lib/libyaml-cpp.dylib
-- Found NetCDF library:     /opt/homebrew/lib/libnetcdf-cxx4.dylib
-- Found Eigen version:      3.4.0
-- Found Boost version:      1.82.0
-- Found Mongocxx version:   3.10.2
-- Found C++ compiler:       AppleClang 15.0.0.15000309
-- Found Git: /usr/bin/git (found version "2.39.3 (Apple Git-146)")
-- Git branch tag:           untagged-3162f8796e5ae38581942e760cf504870dea5eb1
-- Git branch:               main
-- CMAKE_CURRENT_BINARY_DIR: /Users/joannakanska/ginan/src/build
-- CMAKE_CURRENT_SOURCE_DIR: /Users/joannakanska/ginan/src
-- CMAKE_CXX_FLAGS:          -Wno-shift-overflow -Wno-string-concatenation -std=c++2a -fpermissive -fno-omit-frame-pointer -Wall -Wno-write-strings -Wno-deprecated-declarations -Wno-narrowing -Wno-unused-but-set-variable -Wno-sign-compare -Wno-unused-variable -Wno-switch -Wno-dangling-else -Wno-misleading-indentation -Wno-extern-c-compat -Wno-format-zero-length -Wno-array-bounds -O3 -Xclang -fopenmp -pthread
-- Configuring done (2.6s)
-- Generating done (0.1s)
-- Build files have been written to: /Users/joannakanska/ginan/src/build

jkanska23, 2024-09-20:
It works! Thank you very much!

Best regards,
Joanna

closed 2024-09-20 by seballgeyer