GeoscienceAustralia/ginan #109: Receiver clock jump handling

ourairquality, 2024-10-06:
**Describe the bug**

Query about clock jump handling, does Ginan support clock corrections that introduce jumps in the pseudo range and carrier phase observations, or does it require clock steering, or might it want the observations smoothed with the adjustment in the epoch times?

**To Reproduce**

Running an example based on the ppp_example script, and using observations from a receiver that can adjust its clock by jumping the pseudo range and carrier phase.

Here is what happens at the clock jump, the receiver has adjusted its time by 1ms to keep within 1ms, but the REC_CLOCK adjustment is not 1ms (give or take the drift) and that appears to mess up the solution.

Processing G04 L5Q P: 
----------------------------------------------------
Measurement for  CODE_MEAS	 G04	LOG_	   25 L5Q

 OBSERVED                +22572681.0550 ~ 3.254e-01 -> 22572681.0550
 RANGE                   -22559680.3567 ~ 0.000e+00 ->    13000.6983
 REC_CLOCK                 -147721.3078 ~ Estimated ->  -134720.6095
 SAT_CLOCK                 +135877.3050 ~ 0.000e+00 ->     1156.6955

Processing G04 L5Q P: 
----------------------------------------------------
Measurement for  CODE_MEAS	 G04	LOG_	   25 L5Q

 OBSERVED                +22261160.8060 ~ 3.225e-01 -> 22261160.8060
 RANGE                   -22546829.9098 ~ 0.000e+00 ->  -285669.1038
 REC_CLOCK                 -148850.4165 ~ Estimated ->  -434519.5202
 SAT_CLOCK                 +135877.3689 ~ 0.000e+00 ->  -298642.1513

**Expected behavior**

Could the the jump in the pseudo range be detected and the time step estimated.

Alternatively does Ginan work with the adjustment absorbed into the epoch times, can it interpolate from that?

Or might Ginan expect the receiver to use clock steering, which might be fine in an open sky, but might no be great otherwise?

Could not find anything in the documentation about this matter.

**Configuration files**

Based on the ppp_example script.

Might it be significant that it was necessary to add "adjust_rec_clocks_by_spp: false"?

**Operating environment (please complete the following information):**

Using the docker v3.1.0 version on Linux

Btw the trace file "pea_version: " entry is blank.

polytopes-design, 2024-10-06:
Hi,
The `adjust_rec_clocks_by_spp` configuration is what enables the handling of 1ms clock jumps, so you will need to re-enable it.
In fact, it should handle receiver clock jumps of any magnitude.

The receiver clock estimated by SPP is compared with the previous SPP value to determine jump sizes. Before PPP is performed a state transition is performed to add or remove the jump from the estimated clock state. 
You should be able to see the adjustments in the receiver trace files as `Adjusting XXXX clock by 1.2345` (this will be in meters, not ms)

Ginan never adds 'corrections' to observations for clock jumps or cycle-slips etc - it adjusts the estimated state so they reflect the new values of the clock/ambiguity - the clocks and ambiguities have actually changed.

There is another config that can be enabled in addition to the `adjust_rec_clocks_by_spp` config: `adjust_clocks_for_jumps_only` which restricts the adjustments to be multiples of half-millisecond values.

I have noticed that there may be issues with non-gps clock jumps, and should hopefully have a fix for that this week.

ourairquality, 2024-10-06:
Okay, thank you.

closed 2024-10-06 by ourairquality