GeoscienceAustralia/ginan #113: PPP_Example works, but result is not sufficient

jiargei, 2024-10-22:
Dear GINAN-Team!
I tried to verify the result of **ppp_example.yaml**. I thought with the given example it should be possible to generate a solid PPP-solution. But after the ppp example was done, the last line in file exampleConfigs/outputs/ppp_example_main/ALIC_ppp_example_main_201919900.POS shows 

`
2019-07-18T23:59:30.000 2019.547943683 -4052052.72489 4212835.97729 -2545104.60870   0.34134   0.34134   0.34134  -0.000   0.000  -0.000  -23.6701103592  133.8855216014   603.23992     0.00172     0.00654    -0.00896   0.34134   0.34134   0.34135   0.000  -0.000   0.000 ginan
`
As the description explains, the sigma for x/y/z is 0.34134 (meters, i guess) which is not a result expected from such a long time (01D_30S.rnx)

What did I wrong or why is the sigma so high? I expected a sigma of 0.006 or something. Would be nice to get this correct. 

Kind regards

seballgeyer, 2024-10-22:
Thank you for your detailed report regarding the PPP solution generated from ppp_example.yaml. It appears that the high sigma values for X, Y, and Z (0.34134 meters) are not meeting your expectations, especially after processing data over a long duration.

The issue arises because the configuration file was modified to test a new feature but was not reverted to its original settings. Specifically, the ANT_DELTA parameter in the estimation section is still enabled. In static positioning, the position estimates and the ANT_DELTA are highly correlated, which can lead to inaccurate results like the ones you’re experiencing.

To resolve this, please turn off the ANT_DELTA parameter in the estimation part of your configuration file. After making this change, you should see a significant improvement in the sigma values,  reducing them (in my test) to around 0.00141, 0.00145, and 0.00084 for X, Y, and Z, respectively. If you have any further questions or need assistance with the configuration, feel free to reach out!

Kind regards,
Sebastien

jiargei, 2024-10-23:
Dear Sebastien!

Thanks, setting the ant_delta to false it works like a charme now! 

It works on Fedora40 and RockyLinux8. 

Thank you very much!

Kind regards,
Jürgen F.