GeoscienceAustralia/ginan #119: Error on example rt_ppp_example.yaml

jiargei, 2024-10-28:
**Describe the bug**
I tried to run the example rt_ppp_example.yaml but it fails. I changed the input sources (other CORS and BCEP).

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'exampleConfigs'
2. Wait until "Synced X receivers"
4. See error

**Expected behavior**
Should start processing stations.

**Configuration files**

[pea_output.txt](https://github.com/user-attachments/files/17549485/pea_output.txt)

```
pea: malloc.c:2617: sysmalloc: Assertion `(old_top == initial_top (av) && old_size == 0) || ((unsigned long) (old_size) >= MINSIZE && prev_inuse (old_top) && ((unsigned long) old_end & (pagesize - 1)) == 0)' failed.
Aborted (core dumped)
```

**Operating environment (please complete the following information):**

- Ginan branch:     main
- Ginan version:    untagged-3162f8796e5ae38581942e760cf504870dea5eb1-dirty
- Commit date:      Mon Sep 2 05:32:22 2024
- Operating system: "Ubuntu 22.04.5 LTS"
- Compiler version: GNU 11.4.0
- Boost version:    1.74.0
- Eigen version:    3.4.0
- Mongocxx version: 3.6.7

These may be found in the 'Compilation details' section of the pea's output.

**Additional context**
Add any other context about the problem here.

smcclusky, 2024-10-28:
Can you create and provide us with a backtrace using gdb on the core file that was produced by that crash (core dump) please.
We would like to be able to see exactly which function and line of code produced it.

smcclusky, 2024-10-28:
Thanks for the comment. 
We are aware of the slow memory leak issue you raised. We don't strictly consider this a "leak", but believe the slow increase in memory usage arises due to the map/s that contain the SV ephemeris information, slowly growing in size with time.
We will take another look at this issue and see if there is a way that these slow increase in memory usage can be mitigated.

jiargei, 2024-10-29:
Hej Sebastien!

Thanks for the update, I switched to the develop-weekly branch. Unfortunately the sigma values are now higher than before. Now it is more than 500 meters, it was ~5 meters.

Kind regards, Jürgen

seballgeyer, 2024-10-29:
Hi Jürgen, 

Yes, I just realized this too. I am looking at it now. 

Sebastien

jiargei, 2024-10-29:
Hej Sebastien!

I switched back to branch release-v3.1.0. The sigma values are now in sub-meter range, but it takes several hours to get there...

```
+STATES/PPP
*	 2024-10-29 08:59:35	             REC_POS	 AMST	   	      0	  4122998.5403211	       0.36205304	     0.05485011	                 	0x117c8ed0	                                        
*	 2024-10-29 08:59:35	             REC_POS	 AMST	   	      1	  1094890.4266538	       0.42046421	     0.30902426	                 	0x117c8ed0	                                        
*	 2024-10-29 08:59:35	             REC_POS	 AMST	   	      2	  4726256.8256289	       0.25640758	     0.03903920	                 	0x117c8ed0	                                        
*	 2024-10-29 08:59:35	             REC_POS	 MRDF	   	      0	  4123612.2445520	       0.36277859	     0.06034817	                 	0x10ece710	                                        
*	 2024-10-29 08:59:35	             REC_POS	 MRDF	   	      1	  1186651.4980470	       0.41900771	     0.31457385	                 	0x10ece710	                                        
*	 2024-10-29 08:59:35	             REC_POS	 MRDF	   	      2	  4703895.7527488	       0.25676796	     0.03884333	                 	0x10ece710	                                        
*	 2024-10-29 08:59:35	             REC_POS	 NSDL	   	      0	  4096426.1308565	       0.36633576	     0.05708903	                 	0x1110af80	                                        
*	 2024-10-29 08:59:35	             REC_POS	 NSDL	   	      1	  1239622.1316339	       0.42082199	     0.30886839	                 	0x1110af80	                                        
*	 2024-10-29 08:59:35	             REC_POS	 NSDL	   	      2	  4713482.4674074	       0.25934142	     0.03881486	                 	0x1110af80	                                        
*	 2024-10-29 08:59:35	             REC_POS	 STPO	   	      0	  4101619.4977111	       0.36261136	     0.05509162	                 	0x11347b20	                                        
*	 2024-10-29 08:59:35	             REC_POS	 STPO	   	      1	  1147812.7175879	       0.42000558	     0.31016811	                 	0x11347b20	                                        
*	 2024-10-29 08:59:35	             REC_POS	 STPO	   	      2	  4732251.5544407	       0.25641895	     0.03822631	                 	0x11347b20	                                        
*	 2024-10-29 08:59:35	             REC_POS	 WIEN	   	      0	  4085136.0975974	       0.36311141	     0.05655818	                 	0x115844f0	                                        
*	 2024-10-29 08:59:35	             REC_POS	 WIEN	   	      1	  1200313.3849334	       0.41936398	     0.31121022	                 	0x115844f0	                                        
*	 2024-10-29 08:59:35	             REC_POS	 WIEN	   	      2	  4733345.0473014	       0.25650217	     0.03837182	                 	0x115844f0	                                        
-STATES/PPP
```

Just wanted to notice you about that, it might help finding the issue.. in Postprocessing the PPP-works really well!

Kind regards, Jürgen

closed 2025-02-03 by seballgeyer