GeoscienceAustralia/ginan #124: Some questions about POD and the use of pod_example.yaml

newton-fly, 2024-11-06:
**Describe the bug**
First of all, thank you for selflessly open sourcing your code. Recently I just started to learn how to use Ginan, and when I tried to use pod_example.yaml, I got this result, it didn't produce anything,not what I expected. Is my command wrong, or am I using the wrong method, or am I missing some file?I would appreciate it if you could tell me what I should do or how to use pod_example.yaml

**Expected behavior**
I want to be able to implement the output of sp3 and clk files

**Screenshots**
[image]

**Operating environment (please complete the following information):**
- OS: Windows11+docker
- Version of Ginan:v3.1.0

These may be found in the 'Compilation details' section of the pea's output.

**Additional context**
If possible, could you please tell me where to modify the configuration file if I want to add satellites of other systems?

smcclusky, 2024-11-06:
Looks like you may have not started the pea with the configuration file correctly.
The command line should be: `../bin/pea --config pod_example.yaml`
Let ma know if that works for you?

newton-fly, 2024-11-06:
Thank you for your response, but unfortunately, a new error has occurred. The `boxwing.yaml `file I downloaded with the command`./getProducts.sh`
[image]

smcclusky, 2024-11-06:
Did you download the required example products and data files? 
The instructions are in the README.md

newton-fly, 2024-11-07:
Yes, I have re-downloaded the data and products in all the examples, but the error is still there

seballgeyer, 2024-11-08:
Hello,

I tried running the pod_example.yaml using the `gnssanalysis/ginan:v3.1.0` Docker image, and it worked successfully for me.

It sounds like the issue might be related to specific steps or files you may not have downloaded or configured properly. One potential difference in my setup is that I downloaded the required files (e.g., `getProducts.sh`, `getData.sh`) directly within the Docker container. This could be a critical step that might be missing in your case.

Could you please provide more detailed steps of what you have done so far? Specifically, if you can share the commands you used, the files you downloaded, and any other configurations, it will be easier to pinpoint where things might be going wrong. I recommend double-checking that all necessary files are properly downloaded within the Docker container and confirming the file paths in your configuration.

Looking forward to your response!

newton-fly, 2024-11-08:
Thanks for your reply, I have found the problem and run it successfully. The reason is that I didn't set the relative path in `pod_example.yaml` correctly, which is really a small error that shouldn't happen. I'm sorry to have wasted your time

closed 2024-11-08 by seballgeyer