GeoscienceAustralia/ginan #125: tropSAAS height and temperature and guard return

ourairquality, 2024-11-12:
**Describe the bug**

Looking at the code for tropSAAS, which uses a temperature vs height of  `temp = 15 - 6.5E-3 * hgt + ZEROC` and also guards the height to between 20000 and -100m, might this guard range be too high for this particular equation, and what should be done for heights < 0.

From this text https://www.eoas.ubc.ca/books/Practical_Meteorology/prmet102/Ch01-atmos-v102b.pdf page 12, eqn 1.16, which matches the above equation, itqualifies this equation as applying up to a height of 11km, and if heights above this a used then the temp does go below 216.6 so it does appear to have that limited range. Perhaps this function should also guard against heights over 11km.

What does a height < 0 mean in this context, would that be the ellipsoid height, and how would that temperature equation be interpreted in that context, might it be better to clip the height at 0 for this temperature equation and/or for dryZTD and dryMap.

Also a possible nit in this function, if the guards fails then it executes `dryMap = 0; wetMap = 0; dryMap = 0; wetMap = 0;` and might that have been a typo and been intended to be `dryZTD = 0; wetZTD = 0; dryMap = 0; wetMap = 0;`

smcclusky, 2024-11-13:
Thanks for this feedback, we will fix the tropSASS function to fix the initialisation issue and generalise the model to include heights H from -1000 m to +51,000m

ourairquality, 2024-11-13:
If there is an issue with the guard return path then tropSBAS() might have the same issue, and some of the other models just return 0 without setting any of the outputs.

smcclusky, 2024-11-13:
Thanks, we will take a look at this function too.