GeoscienceAustralia/ginan #126: rec.antDelta ordering ENU?

ourairquality, 2024-11-19:
**Describe the bug**

antDelta is defined in src/cpp/common/receiver.hpp as ENU
`	VectorEnu					antDelta;							///< antenna delta {rov_e,rov_n,rov_u}`

Looking at the code in src/cpp/common/rinexObsWrite.cpp and might the output order here be UNE and not H/E/N, and might the output order for elements 1 and 0 need swapping?
	```
tracepdeex(0, rinexStream, "%14.4f%14.4f%14.4f%-18s%-20s\n",
		rec.antDelta[2],
		rec.antDelta[1],
		rec.antDelta[0],
		"",
		"ANTENNA: DELTA H/E/N");

```

antDelta appears to translate directly from models:eccentricity:offset and is this intended to be ENU? Adding '(ENU)' in a comments for this option in the example files might be helpful.

ourairquality, 2024-12-10:
Appears to be corrected now.

closed 2024-12-10 by ourairquality

seballgeyer, 2024-12-10:
Hi, 
My apologies for not keeping you up to date. 
Yes, we did push a patch for it last Friday.