GeoscienceAustralia/ginan #131: can't run GinanEDA

Ali-Fallahi, 2025-01-20:
Hey, dear Ginan support.
I installed Jinan using docker with this code:
`docker run -it -p 5000:5000 -v :/data gnssanalysis/ginan:v3.0.0 bash
`
I successfully downloaded example inputs and also ran MongoDB in the background using these codes:
```
mkdir -p /var/log/mongodb
touch /var/log/mongodb/mongod.log
chmod -R 755 /var/log/mongodb

mkdir -p /data/db/path
chown -R mongodb:mongodb /data/db/path

mongod --dbpath /data/db --logpath /var/log/mongodb/mongod.log --fork --bind_ip 127.0.0.1 --port 27017
```

after all of that I ran pea --config ppp_example.yaml and successfully got all trace files and GPS files

now I want to run GinanEDA but unfortunately, it does not appear in my browser(I use localhost).
I run the index.py and this code appears:
root@b7d249c42820:/ginan/scripts/GinanEDA# python3 index.py 
Serving Flask app 'index'
Debug mode: off
WARNING: This is a development server. Do not use it in a production deployment. Use a production WSGI server instead.
Running on http://127.0.0.1:5000
Press CTRL+C to quit

seballgeyer, 2025-01-27:
Hello

You’ve successfully set up Ginan in a Docker container and are trying to run GinanEDA using the `index.py` script. However, you're finding that the web interface isn’t accessible from your browser (`localhost:5000`), even though the Flask server is running inside the container. This is because the Flask app is bound to `127.0.0.1`, which restricts access to just inside the container.

By default, when Flask runs on `127.0.0.1`, it’s only accessible from within the container itself. This prevents you from reaching the web interface from your host machine (your Windows system). To make the GinanEDA interface accessible from your browser, you need to make the Flask server listen on all network interfaces, not just `127.0.0.1`.

To fix this, you can modify the command you’re using to run Flask inside the container. Use the following command to bind Flask to 0.0.0.0, which makes the app accessible from outside the container (including your Windows host machine):
```bash
FLASK_APP=index.py FLASK_RUN_HOST=0.0.0.0 flask run --host=0.0.0.0 --port=5000
```

This will allow you to access the GinanEDA web interface at `http://localhost:5000` on your browser, so you can interact with the tool and analyze your GNSS data.

You can run this command interactively inside the Docker container, and it will make the tool accessible without modifying any internal Flask code. This should solve your issue and allow you to fully use the EDA tool.

Ali-Fallahi, 2025-02-01:
thanks for the support.
I successfully ran GinanEDA on my localhost.

closed 2025-02-03 by seballgeyer