GeoscienceAustralia/ginan #133: CODE ERP files not recognised as IGS format

ourairquality, 2025-02-10:
**Describe the bug**

The CODE final EPR files have a few extra columns, compared with the regular IGS format, and this causes the line length test used to recognise this format to fail. See erp.c readErp()

`if (  line.size()	<= 127	&& line.size()	>= 106)	readIgsErp(line, erpMap);`

This failure causes it to fall through to  readIersBulletinA() which is a completely different format and nothing is read.

Here's the IGS format https://lists.igs.org/pipermail/igsmail/1998/003315.html but it does not include the addition fields in following example CODE final EPR file. Could not find a definition of the CODE ERP format.

```
version 2
CNES CLS GRGS ERP VALUES 
________________________________________________________________________________________________________
  MJD       Xpole   Ypole   UT1-UTC   LOD    Xsig   Ysig UTsig LODsig Nr Nf Nt  X-RT  Y-RT  X-RTsig Y-RTsig Dpsi  Deps
             E-6"    E-6"  0.1 us  .1us/d    E-6"   E-6" .1 us .1us/d         E-6"  E-6"   E-6"    E-6" E-6"  E-6"   
60712.75   90427  310035   456290   -3347      1      1     0     0   0  0  0 -2641   399     0     0      0      0
60713.75   87786  310434   460393   -4860      0      0     0     0   0  0  0 -2641   399     0     0      0      0

```

Hacking the line size test above allowed it to proceed to read in this CODE ERP file.

Detection of the IGS format by checking for "version 2" or "VERSION 2" at the start of the first line might be a suggestion.

seballgeyer, 2025-02-11:
thank you, this bug has been fixed a few days ago, and is going now under review. the fix will be in our weekly push to github on Friday

ourairquality, 2025-02-16:
Thank you, the CODE ERP products now load.

closed 2025-02-16 by ourairquality