GeoscienceAustralia/ginan #134: Possible issue using ERP products, without eop estimation enabled?

ourairquality, 2025-02-10:
For a ppp solution, added an ERP file to the config file under `inputs: erp_files:` and with `estimation_parameters: global_models:  eop:` disabled then the ERP data was not used.

There is a test in `erp.cpp getErp() if (recOpts.eop == false)  return erpv;` which appeared to be causing the code to give up on the search. If this test was disabled then it to proceed to find and use the ERP data.

In `ppp_obs.cpp receiverUducGnss() if (recOpts.eop) pppEopAdjustment(COMMON_PPP_ARGS);` this same flag appears to be used enable estimation.

Should it be possible to use an ERP product without estimation enabled, and is there a way to do so?

seballgeyer, 2025-02-11:
Yes, it is possible to use an ERP product without enabling estimation. To do so, you simply need to enable the relevant option in your configuration file. You can do this by setting 
```yaml
receiver_options:
    global:
        model:
            eop:
                enable: true 
```
in your YAML file. This will allow the ERP data to be used even when estimation is not enabled

ourairquality, 2025-02-16:
Thank you, that got it working. See now that eop estimation requires **both** `receiver_options:global:models:eop:enable:true` and `estimation_parameters:global_models:eop:estimated:true`, and the using an ERP product requires just `receiver_options:global:models:eop:enable:true`.

But the ERP data is an input to a few paths in tideDisp() and these paths are still taken even without the EPR data being returned by getERP(), without recOpts.eop enabled it returns garbage even if an ERP data product file is supplied. Further this option is now even mentioned in the ppp example and would be need to use the EPR product or estimation. Is there any reason to even have this recOpts.eop flag - it seems it could just be removed to resolve some issues here.