GeoscienceAustralia/ginan #135: IERS secular mean pole time scale clarification

ourairquality, 2025-02-10:
Possible trivial nit. In `erp.cpp tideSolidPole()` the time passed is has class `MjDateUt1 `and this is passed to `iersMeanPole()` with the same time class and is then used via `mjdUt1.to_j2000()`.

Looking at the latest IERS documentation, where this model is defined, section _7.1.4 Rotational deformation due to polar motion: Secular polar motion and the pole tide_ it notes _t is the date in years of 365.25 days_ and uses this as _(t - 2000)._ I could not find an article detailing their least square fit that might have clarified the exact time scale used, but looking at some of their other code it might be that the epoch is J2000.0 (which they appear to commonly use) and the time scale is likely without leap seconds from this epoch.

In which case this would appear to be equivalent to (GPST - J2000.0_GPST)  / 86400.0 / 365.25 where J2000.0_GPST is 2000/1/1 11:59:08.816 GPST.

FWIW when I tired this the time was slightly different to that in your implementation of `iersMeanPole()`, around 63 seconds difference (perhaps a coincidence but close to 13 + 19 + 32.184).

Would you happen to know what the expected time scale is intended to be here? Could a code comment spell it out (since your code is a useful reference, thanks). Might passing in UT1 be somewhat convoluted, and might passing in GPST or TT (TDT) or TIA etc be clearer if these are the intended time scale (less a constant offset).

seballgeyer, 2025-02-11:
Thanks for bringing this up. Here’s what I understand from your issue:

You’ve noticed that the tideSolidPole() function passes a MjDateUt1 object to iersMeanPole(), and the time is then used via mjdUt1.to_j2000(). Looking into the latest IERS documentation, it seems that the mean pole equations expects time in terms of J2000.0 (Defined in the glossary as January 1, 2000, 12:00:00 TT), but the function you refer currently uses UT1, which could lead to minor differences. 
We calculate that the UT1 vs TT time system difference (~69 sec) would amount to ~ nano arc second differences in the mean pole correction at the present day. 

To resolve this, I’d suggest a couple of changes:
- Clarify the Time Scale in Code: We will add a comment in the code that explicitly mentions the expected time scale and epoch (i.e., TT and J2000.0). This would provide clarity for others using or modifying the code in the future.
- Deprecate iersMeanPole(): There’s an updated version of iersMeanPole() in orbprop/iers2010.cpp that properly handles time in TT.  We will remove the old version in favor of the newer one to ensure that the time is correctly handled according to the IERS standards.

By implementing these changes, I think we can ensure the code aligns better with the IERS standards and reduce any time-related discrepancies in the polar motion calculations.

ourairquality, 2025-02-16:
Yes it was a trivial matter, but thank you for clarifying this.

closed 2025-02-16 by ourairquality